ev3dev/ev3dev-lang #32: Cannot detect color sensor

ftheriault, 2015-01-08:
Hi all,

I'm quite new using ev3dev, so please don't be too hard on me.

I have installed ev3dev and executed the CPP samples. The ev3dev-lang-test program seems to detect properly my motors and my touch sensor, though it doesn't detect my EV3 color sensor. I also tried a NXT color sensor, without more success...

Am I the only one having this issue? 

Thanks!

dlech, 2015-01-08:
There is no driver for the NXT Color sensor yet, so no, you are not the only one. :wink:

The EV3 color sensor should work though. What kernel version are you using?

ftheriault, 2015-01-08:
Thanks for the quick reply!

I am using the following image version : ev3dev-jessie-2014-12-01.img

Kernel is :
Linux ev3dev 3.16.1-7-ev3dev #2 PREEMPT Tue Nov 25 11:24:38 CST 2014 armv5tejl GNU/Linux

dlech, 2015-01-08:
There are some breaking changes in the more recent kernels, but I think that one should work. You can see what is is `/sys/class/msensor` to see if the sensor is actually there. If it is there, then then I am not sure what the problem is.

ftheriault, 2015-01-09:
Funny thing is... it is not there :(

I see my touch sensor, but that's it... any clue?

dlech, 2015-01-09:
Have you updated other packages? Namely, what is the version from `dpkg -s ev3dev-base`. If you upgraded this package (`apt-get upgrade`) without upgrading the kernel (`apt-get dist-upgrade`), then this is why it is not working. If you do upgrade the kernel though, there are some breaking changes (see #29) and you will have to fix the cpp library.

ftheriault, 2015-01-09:
Humm... the only thing I did was : 
apt-get install build-essential, for the gcc compiler (I did not know we used CC at the time...). Could it be the problem?

Here is the result of `dpkg -s ev3dev-base`, which I did not upgrade :

Package: ev3dev-base
Status: install ok installed
Priority: required
Section: metapackages
Installed-Size: 37
Maintainer: David Lechner <david@lechnology.com>
Architecture: armel
Version: 8.0
Depends: ev3dev-bluetooth (>= 8.0), ev3dev-console (>= 8.0), ev3dev-flash-kernel (>= 8.0), ev3dev-gadget (>= 8.0), ev3dev-i2c (>= 8.0), ev3dev-leds (>= 8.0), ev3dev-ports (>= 8.0), ev3dev-uart (>= 8.0)
Description: Meta package that installs all of the required ev3dev packages.
Homepage: http://www.ev3dev.org

ftheriault, 2015-01-09:
Checking my command history, I see that at one point, I tried to install ev3dev-c (see in4lio project, on github). If you think this can be the problem, I'll just recreate the image...

dlech, 2015-01-09:
[ev3dev-base 8.0](https://github.com/ev3dev/ev3dev-base/blob/ev3dev-jessie/debian/changelog) is for kernel v3.16.1.-8-ev3dev and higher which is why UART sensors are not working. (So, you did upgrade even if not intentionally.)

So, your options are: you can recreate the image and use the old versions or you can upgrade the kernel and help migrate the cpp library to the new kernel version.

ftheriault, 2015-01-09:
Thanks David, I really appreciate the time that you took to help. Since I am way too novice at this point, I'll just revert a fresh image.

closed 2015-01-09 by ftheriault

dlech, 2015-01-09:
You are welcome! And the changes are just changing strings for things that were renamed, so even a novice could do it if you are feeling adventurous.