ev3dev/ev3dev-lang #33: Inconsistent handling of 'command' attributes

fdetro, 2015-01-11:
I'm currently adjusting my cpp code to the autogen stuff and see some inconsistencies with the ``command``attributes that break my test code:

* class ``dcMotor`` has attributes ``command``and ``commands``, but ``command``is read-only
* class ``servoMotor`` has only attribute ``command``, here read-write

Is this a bug in the spec or can @dlech comment on this?

dlech, 2015-01-11:
dc-motor class command attribute is read/write. [docs](http://www.ev3dev.org/docs/drivers/dc-motor-class/)

The servo-motor class needs to have a `commands` attribute added, but doesn't have one yet.

In the future, I may change the `command` attribute to be write-only like the lego-sensor class and have a separate `status` attribute (this applies to the tacho-motor class as well).

dlech, 2015-01-11:
In case it is not clear, all of my remarks above are about the kernel drivers themselves, not the spec. So, yes `command` is read-only is an error in the spec.

WasabiFan, 2015-01-11:
Here are the relevant pages for reference:
- http://www.ev3dev.org/docs/drivers/lego-sensor-class/
- http://www.ev3dev.org/docs/drivers/dc-motor-class/
- http://www.ev3dev.org/docs/drivers/servo-motor-class/

WasabiFan, 2015-01-11:
@dlech the command attribute is write-only in [our spec](https://github.com/ev3dev/ev3dev-lang/blob/develop/wrapper-specification.md#direct-attribute-mappings-1), not read only. But either way it's not how it is in the kernel, so I'll change it.

commit 035fda91 (2015-04-17): Fix access modifiers on 'command' attribute in dcmotor (closes #33)
closed 2015-04-17 by WasabiFan