ev3dev/ev3dev-lang #35: bus error when dpkg-reconfigure locales

crowndale1, 2015-01-19:
I got bus error when dpkg-reconfigure locales. I used the default settings - en_US .... Anyone can help me? Thanks.

dlech, 2015-01-19:
This doesn't sound like a problem related to the ev3dev-lang libraries. Better repost at https://github.com/ev3dev/ev3dev/issues. Be sure to read [this](https://github.com/ev3dev/ev3dev/blob/master/CONTRIBUTING.md) before posting too.

crowndale1, 2015-01-22:
Dave,

 

Here is my issue:

 

1) download ev3dev-jessie-2014-10-07 -OK

2) follow the instructions to connect the ev3 to a window PC using USB -OK

3) in Putty, user name: root, password: r00tme - OK

3) in Putty, dpkg-reconfigure tzdata - OK

4) in Putty, dpkg-reconfigure locales - Error

 bus error 

 

Thanks.
 

Date: Mon, 19 Jan 2015 14:12:07 -0800

dlech, 2015-01-22:
Like I said, this is not the correct place to report this problem. See my previous comment.

closed 2015-01-22 by dlech