ev3dev/ev3dev-lang #37: autogen usage

ddemidov, 2015-01-22:
Hi,

I am looking at converting the python bindings to using the autogen scripts. For now I am just trying to run the existing scripts. Unfortunately, the [usage instructions](https://github.com/ev3dev/ev3dev-lang/tree/develop/autogen#usage) do not work for me: https://gist.github.com/ddemidov/3b97a533a149491c7006

Do I need anything else for autogen to work?

WasabiFan, 2015-01-22:
You're the first person (as far as I know) to test this in it's current state, so there may be some issues (sorry!).

Can you try again (pull upstream changes first)? I think the problem was that I hadn't fully removed some bad scripts from the fork of the library that our autogen package tries to install. Because it is installing from a git repo instead of from NPM itself, NPM doesn't automatically run the build scripts, so I had to configure it manually (and the issue was in that custom configuration).

I also made some changes to the autogen readme so that it's more understandable.

ddemidov, 2015-01-23:
This manages to get further, but `npm install` still ends with an error: https://gist.github.com/4164b089de53f82c752e

WasabiFan, 2015-01-25:
@ddemidov Sorry for the late reply, I was very busy and completely forgot to reply to this thread.

I think I have fixed all the issues (and I have tested it on multiple clean installs of the OS). The problem appears to be how `npm` ran my scripts: the module was installing and running the `prepublish` script, which would run the `compile` script, which would try to install some dependencies, which would trigger the `prepublish` again and create an infinite loop. It should be fixed now -- and thanks for your patience!

ddemidov, 2015-01-26:
It works! Thank you, I'll continue my experiments with autogen.

ddemidov, 2015-01-26:
The following command runs fine:
```
$ node autogen.js 
4 files found
Completed processing file "js/extras.ts"
Completed processing file "js/motor.ts"
Completed processing file "js/sensor.ts"
Completed processing file "wrapper-specification.md"
```
If, however, I modify an autogen block inside one of the generated files (for example
~~~{.diff}
diff --git a/extras.ts b/extras.ts
index 8e3338f..29784d0 100644
--- a/extras.ts
+++ b/extras.ts
@@ -3,7 +3,6 @@
 ///<reference path="io.ts" />
 
 //~autogen autogen-header
-    // Sections of the following code were auto-generated based on spec v0.9.2-pre, rev 1. 
 //~autogen
 
 class PowerSupply extends Device {
~~~
),  then `node autogen.js` results in the following error:
```
$ node autogen.js 
4 files found
Completed processing file "js/motor.ts"
Completed processing file "js/sensor.ts"
Completed processing file "wrapper-specification.md"
Error processing file "js/extras.ts": Error: ENOENT, open 'js/extras.ts'
```
It seems that autogen does not overwrite a file if its contents did not change, but fails when there are some changes to apply.

ddemidov, 2015-01-26:
The error above does not happen when `node` is run from the top level directory:
```
$ node autogen/autogen.js 
4 files found
Completed processing file "js/motor.ts"
Completed processing file "js/sensor.ts"
Completed processing file "wrapper-specification.md"
Completed processing file "js/extras.ts"
```
Is that the intended behavior?

closed 2015-01-26 by WasabiFan

WasabiFan, 2015-01-26:
@ddemidov Feel free to reopen this if you come across any more issues.

PR #38 (2015-01-26): Use correct filename in autogen.js when writing results back
This fixes #37
commit a38c7e1c (2015-01-26): Use correct filename when writing results back

This fixes #37