ev3dev/ev3dev-lang #41: JavaScript binding - no sensor name and dp value.

donwojtallo, 2015-01-31:
Creating new sensor using ```new ev3dev.Sensor();``` returns:
```Error: ENOENT, no such file or directory '/sys/class/lego-sensor/sensor1/name'```
It looks like there is no name value anymore.
Here is workaround to make sensor work. In module/bin/index.js ev3dev/nstead of line:
```var typeName = fs.readFileSync(path.join(rootPath, "name")).toString().trim();```
I put this:
```var typeName = '';```

The second error is no dp value:
``` Error: ENOENT, no such file or directory '/sys/class/lego-sensor/sensor1/dp'```
The example script returns this inside ```getFloatValue()``` function. I tried this on gyro sensor.

Also, small thing, the package on ```npm``` repository is outdated, using the sensor shows error which were already fixed on Github.

WasabiFan, 2015-01-31:
@donwojtallo sorry about that; yes, the npm package is outdated. I publish to npm when we do a release on GitHub, which happens when we are finished with our changes across all languages.

So, until the module is updated on npm, I recommend that you clone my "ev3dev-lang-js" repo and follow the instructions in the readme to build from source. Then you can just copy the new "index.js" file that you built to wherever your project is and use that instead for now.

donwojtallo, 2015-01-31:
I am already working on cloned repo and report problems above with that version.

WasabiFan, 2015-02-01:
@donwojtallo Pull my recent changes and try again; I believe I fixed the issue.

Thanks for pointing this out!

donwojtallo, 2015-02-02:
Sorry for delay. I've tested it and now it works!
I've also fixed example file ([see](https://github.com/WasabiFan/ev3dev-lang-js/pull/1)).
Please, update npm package too. I think it's important to keep this package up-to-date.
Thank you!

WasabiFan, 2015-02-04:
@donwojtallo Can we close this?

donwojtallo, 2015-02-04:
I thought You will update npm package before closing this. If You believe that it's not related to this issue, I am closing this right now. Sorry for misunderstanding.

closed 2015-02-04 by donwojtallo

WasabiFan, 2015-02-04:
Because we use a single version number on most of our bindings, I can't update the npm package without becoming out-of-sync with the other libraries. I may be able to publish a "beta" version without updating the version number; I'll see what I can do and post back here if I get it published.