ev3dev/ev3dev-lang #43: JavaScript binding - test-motor-sensor.js does not find '/sys/class/lego-sensor/'

AleksiAhtiainen, 2015-02-10:
Hi,

I'm trying to get node.js bindings working on ev3dev. I found out that examples/test-motor-sensor.js does not work. See below. Has the folder been renamed or the interface to motor sensors changed? I am assuming this example should work with the large/medium motors EV3 motors.

I am using latest self-compiled js bindings.

```text
alah@ev3dev:~/work/ev3-js-tests$ nodejs test-motor-sensor.js
Motor --------------
dutyCycle: 0
dutyCycleSp: 0
portName: outA
position: 0
positionSp: 0
pulsesPerSecond: 0
pulsesPerSecondSp: 0
rampDownSp: 0
rampUpSp: 0
regulationMode: off
run: 0
runMode: forever
speedRegulationP: 1000
state: idle
stopMode: coast
stopModes: coast,brake,hold
timeSp: 0
type: tacho
Setting motor properties...
Running motor...
--------------------
Sensor -------------
{ [Error: ENOENT, no such file or directory '/sys/class/lego-sensor/']
  errno: 34,
  code: 'ENOENT',
  path: '/sys/class/lego-sensor/',
  syscall: 'readdir' }

Error: You must be connected to a device before you can read from it.
    at Sensor.Device.readProperty (/home/alah/work/index.js:98:19)
    at Sensor.Device.readString (/home/alah/work/index.js:87:26)
    at Sensor.Device.getString (/home/alah/work/index.js:94:21)
    at Sensor.Object.defineProperty.get [as portName] (/home/alah/work/index.js:768:25)
    at i (/home/alah/work/ev3-js-tests/test-motor-sensor.js:31:38)
    at Array.forEach (native)
    at Object.<anonymous> (/home/alah/work/ev3-js-tests/test-motor-sensor.js:30:3)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
```

dlech, 2015-02-10:
What kernel version are you using? (`uname -r`)

AleksiAhtiainen, 2015-02-10:
Sorry, forgot to mention that. I just upgraded everything via apt-get update + upgrade + dist-upgrade last night, uname -r:
3.16.1-7-ev3dev

dlech, 2015-02-10:
It did not upgrade the kernel. Try `apt-get install linux-image-ev3dev`. The latest version is `3.16.7-ckt4-ev3dev1`.

BertLindeman, 2015-02-10:
> I just upgraded everything via apt-get update + upgrade + dist-upgrade last night, uname -r:
3.16.1-7-ev3dev

Did you do a reboot after the kernel upgrade?

dlech, 2015-02-10:
Yes, you need to reboot after kernel upgrade.

AleksiAhtiainen, 2015-02-11:
Yes, I rebooted. 

I started from the ev3dev-jessie-2014-12-01 image.

I think I also saw 3.16.7-ckt4-ev3dev1 mentioned during the about 3 hour upgrade process. At one point I was asked about some kernel config file whether I want to keep some local changes or not, and after checking out the diff I decided to keep the local changes.

According to sudo apt-get install linux-image-ev3dev "linux-image-ev3dev is already the newest version."

dlech, 2015-02-11:
I guess I am a bit overdue for a new test image if it takes that long for the initial package updates.

Just to confirm the `linux-image-ev3dev` version, running `dpkg -s linux-image-ev3dev` should look like this:

```
Package: linux-image-ev3dev
Status: install ok installed
Priority: optional
Section: kernel
Installed-Size: 26
Maintainer: David Lechner <david@lechnology.com>
Architecture: armel
Source: linux-latest-ev3dev
Version: 16
Provides: linux-latest-modules-3.16.7-ckt4-ev3dev1
Depends: linux-image-3.16.7-ckt4-ev3dev1, rtl8188eu-modules-3.16.7-ckt4-ev3dev1, rtl8192cu-modules-3.16.7-ckt4-ev3dev1
Description: Linux kernel for ev3dev (meta-package)
 This package depends on the latest Linux kernel and modules from the ev3dev
 project for use on the LEGO Mindstorms EV3.
```

Then, if you run `sudo flash-kernel`, it should say:

```
flash-kernel: installing version 3.16.7-ckt4-ev3dev1Generating kernel u-boot image... done.
Taking backup of uImage.
Installing new uImage.
Generating initramfs u-boot image... done.
Taking backup of uInitrd.
Installing new uInitrd.
Generating boot script u-boot image... done.
Taking backup of boot.scr.
Installing new boot.scr.
```

If the kernel version is correct here, then after a reboot, you will have the up-to-date kernel.

AleksiAhtiainen, 2015-02-11:
```dpkg -s linux-image-ev3dev``` gives the same output as in your sample.

During apt-get upgrade, ```flash-kernel``` is interrupted with a warning, and I probably didn't notice the error message yesterday:
```text
Installing new boot.scr.
W: Operation was interrupted before it could finish
```

But also: Running flash-kernel manually doesn't give any warning (same output as yours, except for some locale complaints), but I still don't have the new kernel after reboot:

```
root@ev3dev:/boot# flash-kernel
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LC_CTYPE = "UTF-8",
	LANG = "en_US.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to a fallback locale ("en_US.UTF-8").
	LC_ALL = (unset),
	LC_CTYPE = "UTF-8",
	LANG = "en_US.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to a fallback locale ("en_US.UTF-8").
flash-kernel: installing version 3.16.7-ckt4-ev3dev1
Generating kernel u-boot image... done.
Taking backup of uImage.
Installing new uImage.
Generating initramfs u-boot image... done.
Taking backup of uInitrd.
Installing new uInitrd.
Generating boot script u-boot image... done.
Taking backup of boot.scr.
Installing new boot.scr.
```

```
root@ev3dev:/boot# reboot
...
alah@ev3dev:~$ uname -r
3.16.1-7-ev3dev
```

AleksiAhtiainen, 2015-02-11:
Even when running ```flash-kernel``` separately, it runs surprisingly fast. So I doubt it is doing anything. ```Taking backup of boot.scr.``` and ```Installing new boot.scr.``` don't seem to take any time at all.

No error messages are printed, though.

AleksiAhtiainen, 2015-02-11:
Thank you for all the help so far!

It seems like the -ckt4 just does not load for me, and the system loads the older kernel from backup.

Is there any way to debug the reason for that?

I did try purging and reinstalling the ```linux-image-3.16.7-ckt4-ev3dev1```, but it does not help. Booting without WLAN USB adapter connected does not help. I am running from a 32GB SD card, with the filesystem grown to the full size, could that be the reason?

Is there some easy way to try downgrading to "ckt3" or some other bit older kernel version?

dlech, 2015-02-11:
Someone else reported this problem today in ev3dev/ev3dev#267. Since the problem is a general ev3dev problem and not specific to ev3dev-lang, I am continuing the discussion there and suggest that this issue be closed as invalid since there is not actually a problem with ev3dev-lang or the javascript bindings. Please see ev3dev/ev3dev#267 for why I think that the kernel upgrade is not taking effect.

AleksiAhtiainen, 2015-02-13:
The solution at https://github.com/ev3dev/ev3dev/issues/267 worked for me.

Now the JS example also works as expected. I first assumed it is reading some motor sensor, and not from the sensor inputs. The example required me to connect some sensor to one of the input ports in addition to the motor in one of the output ports.

closed 2015-02-13 by WasabiFan