ev3dev/ev3dev-lang #46: add a test ht-nxt-angle and accumulated angle

cho934, 2015-03-21:
Hello,

I've just tested ev3dev-lang-demo and ev3dev-lang-test connecting a Hitechnic angle sansor (http://www.ev3dev.org/docs/sensors/hitechnic-nxt-angle-sensor/)
The demo executable recognized it as <ht-nxt-angle>, but without accumulated angle function. Is it possible to add it ?

I tried to add the line in ev3dev-lang-test:
test_sensor<i2c_sensor>("ht-nxt-angle");
So how can I add this device in ev3dev-lang-test ?

dlech, 2015-03-21:
It looks like the test only uses the current mode of the sensor, so you could manually set the mode before running the test (as in `echo ACCEL > /sys/class/lego-sensor/sensor0/mode`).

I also notice that the accumulate mode is currently called `ACCEL`, which is a mistake, so this name will be changing in the next kernel release.

cho934, 2015-03-21:
I received the message "No ht-nxt-angle sensor found".
Is the name "ht-nxt-angle" correct ?

So I tested again ev3dev-lang-demo with the angle sensor and the mode ACCEL.
It seems the value are completely incorrect. Below are the results of the command "monitor":
*** ht-nxt-angle (ACCEL) actions ***

(s)how modes
(c)hange mode
(v)alue
(m)onitor

(b)ack

Choice: m
1107623936 deg
1124401152 deg
1141178368 deg
1124401152 deg
1074069504 deg
1023737856 deg
956628992 deg
906297344 deg
872742912 deg
805634048 deg
788856832 deg
772079616 deg
788856832 deg
805634048 deg
906297344 deg
1225064448 deg
1476722688 deg
1829044224 deg
-2096824320 deg
-1761280000 deg
-1476067328 deg
-972750848 deg
-838533120 deg
-788201472 deg
-804978688 deg
-788201472 deg
-804978688 deg
-788201472 deg

An idea of the problem ? in the kernel also ?

dlech, 2015-03-21:
What is your kernel version?

What is output of `cat /sys/class/lego-sensor/sensor0/driver_name`? This is the "correct" name and it should match the name on the documentation page. It appears from the output of the demo, `ht-nxt-sensor` is correct.

I think the "No ht-nxt-angle sensor found" error is just because the cpp bindings have not been updated for recent kernel versions.

The other issues are kernel issues. For example if the numbers don't look correct, it could be a endianness problem. You should open a new issue in the main ev3dev issues for this.

cho934, 2015-03-21:
Thank you David for your help.

user@ev3dev:~$cat /sys/class/lego-sensor/sensor0/driver_name
ht-nxt-angle

I use the last image ev3dev-jessie-2015-02-24, and the kernel is the following:

user@ev3dev:~$ uname -a
Linux ev3dev 3.16.7-ckt6-ev3dev1 #2 PREEMPT Sun Feb 22 16:07:24 CST 2015 armv5tejl GNU/Linux

dlech, 2015-03-21:
That all looks good, so there is probably some bug in ev3dev-lang in the ev3dev-lang-test and additionally a bug in the kernel driver related to the accumulated angle.

ddemidov, 2015-05-12:
@cho934, is this still a problem with the latest kernel/C++ bindings?

closed 2015-12-19 by WasabiFan