ev3dev/ev3dev-lang #49: Python demos

Svetomech, 2015-04-12:
They simply doesn't work. "Can't import PIL package" or something. I tried installing it through pip, but it says it is incompatible with ev3dev.
Tried installing Pillow, failed (memory error at 99%).
So, what do I do? Or does this mean current state of Python binding is not usable?

ddemidov, 2015-04-12:
I installed PIL with `apt-get install python-pil`, and it works for me. Can you try that? Do the rest of the demos work for you as is?

Svetomech, 2015-04-12:
Well, not really. I followed your instructions in https://github.com/ddemidov/ev3dev-lang-python:
> apt-get install libboost-python-dev python-setuptools python-pil
> easy_install python-ev3dev
Then transferred 'demo' folder to the device using scp and tried launching .py files. None of them work.

ddemidov, 2015-04-12:
The demos assume that certain devices are connected, and may fail otherwise. Can you try to connect the touch sensor to your ev3 and try the following?
```python
>>> from ev3dev import *
>>> ts = touch_sensor()
>>> ts.connected
True
>>> ts.port_name
'in1'
>>> ts.value()
0
```

Svetomech, 2015-04-12:
Works as intended, duh.

ddemidov, 2015-04-12:
If you connect two large motors to ports B and C, an infrared and a touch sensor (to any ports), then remote-control.py should work for you. If it still does not, can you post the error message here?

ddemidov, 2015-04-28:
May this be closed?

Svetomech, 2015-04-28:
Sorry, forgot about the issue.
Eeyup.

closed 2015-04-28 by WasabiFan