ev3dev/ev3dev-lang #50: vala bindings

dlech, 2015-04-17:
I have merged the vala bindings into the ev3devKit library. So, my question is: do you want to keep it as a part of ev3dev-lang or let it just be its own library?

WasabiFan, 2015-04-17:
You have been taking much better care of them than I ever could have, and are getting good use out of them as they are. So, I think that unless there's a reason that you can think of to attempt to keep the vala binding intertwined with this set of bindings, we should break it off and list it separately (on ev3dev.org and in this repo).

dlech, 2015-04-17:
I wasn't planning on using the autogen stuff anyway, so I think we will just keep them separate.

closed 2015-04-17 by dlech

WasabiFan, 2015-04-17:
Sounds good -- I'll remove the submodule and listings from this repo.