ev3dev/ev3dev-lang #53: Speed/Hold PID property names are incorrect

ddemidov, 2015-04-20:
The [autogen/spec.json](https://github.com/ev3dev/ev3dev-lang/blob/develop/autogen/spec.json) is incorrectly names the [hold](https://github.com/ev3dev/ev3dev-lang/blob/35c4d5fd3c5038c2a447ceb812f35e5643d9de25/autogen/spec.json#L25-L27)/[speed](https://github.com/ev3dev/ev3dev-lang/blob/35c4d5fd3c5038c2a447ceb812f35e5643d9de25/autogen/spec.json#L34-L36) regulation parameters. See the current driver docs [here](https://github.com/ev3dev/lego-linux-drivers/blob/282b76e91932dbc3887691f45432399e782ac18e/motors/tacho_motor_class.c#L103-L110) and [here](https://github.com/ev3dev/lego-linux-drivers/blob/282b76e91932dbc3887691f45432399e782ac18e/motors/tacho_motor_class.c#L149-L156).

ddemidov, 2015-04-20:
Sorry for the noise, I did not look at the system names.

ddemidov, 2015-04-20:
It turns out that the real problem is that the specification states that `hold_pid/*` properties have read access, but reading the properties results in an error:
```
cat /sys/class/tacho-motor/motor0/hold_pid/Kp
cat: /sys/class/tacho-motor/motor0/hold_pid/Kp: Operation not supported
```

WasabiFan, 2015-04-20:
Don't the docs files that you linked to say that they support both read and write?

ddemidov, 2015-04-20:
The docs say so, but the properties are not readable on my ev3. This is probably an error in the driver, so I opened ev3dev/ev3dev#314 as well.

dlech, 2015-04-21:
If an attribute does not have read or write access, you will get a permission error (`-EACCESS` I think) rather than an operation not supported error (`-EOPNOTSUPP`). This is working as indented. Reading and writing the hold_pid parameters for the `legoev3-motor` driver is not implemented, thus the `-EOPNOTSUPP` error. When using the `ms-nxtmmx` driver, these attributes will actually do something.

ddemidov, 2015-04-21:
Ok, thanks for the explanation. This means that currently both ev3dev-lang-test.cpp and ev3dev-lang-demo.cpp shouldbe edited manually after autogeneration (as I have done [here](https://github.com/ev3dev/ev3dev-lang/blob/b2ed3d2f96930d747726e08a236ad8118e201a67/cpp/ev3dev-lang-test.cpp#L91-L95)).

I am thinking of adding a try/catch block to the generated text [here](https://github.com/ev3dev/ev3dev-lang/blob/b2ed3d2f96930d747726e08a236ad8118e201a67/autogen/templates/cpp_generic_report_status.liquid#L4) and outputting something like 'not supported' when this exception is thrown. @WasabiFan, what do you think?

WasabiFan, 2015-04-21:
> I am thinking of adding a try/catch block to the generated text here and outputting something like 'not supported' when this exception is thrown. @WasabiFan, what do you think?

That sounds like it would not only mitigate the errors, but also provide useful information on the status and usage of the motors, so I'm happy with that plan. Just make sure that the message makes it clear that the error isn't actually a problem.

ddemidov, 2015-04-21:
Resolved by #55

closed 2015-04-21 by ddemidov