ev3dev/ev3dev-lang #57: NXTServo 8 channel and ev3dev-lang-demo

cho934, 2015-04-23:
Hello,

Just to confirm before testing servomotors, should I connect NXTServo-V3 above (motor out port) or below (sensor in port) to the ev3brick, because it is notice to not connect to motor port (doing so may damage NXTServo-V3) ?

dlech, 2015-04-23:
The `ms-8ch-servo` device in `/sys/class/lego-sensors/` is the controller itself. And yes the only useful thing there is the battery voltage.

There should also be 8 devices in `/sys/class/servo-motor/`. These are are used to actually control the motors.

I don't know if the demo has anything implemented for the servo-motor class or not.

cho934, 2015-04-23:
I don't see anything.
Thanks for the quick answer. It works like a charm using  /sys/class/servo-motor/

For information, we need to use 2 servos and try to finish to build our secondary robot this week-end. The French cup of robotic will occure in mid-May, and I will promote ev3dev ;)

ddemidov, 2015-04-24:
There is a servo motor test in [ev3dev-lang-test.cpp](https://github.com/ev3dev/ev3dev-lang/blob/develop/cpp/ev3dev-lang-test.cpp#L230-L274). It does not do much, just dumps the motor's properties. Does it work for you?

cho934, 2015-04-24:
Yes, it works and it shows information about connected servomotors. I'm going to work now on our program in C++ using ev3dev.cpp and I will check if I can add a test to move servos in ev3dev-lang-demo.

cho934, 2015-04-26:
I did'nt succeed to drive several servos using ev3dev.h. The code from ev3dev-lang-test just detected only the first servo.
Have you got a small test to show me how to control several servos using ev3dev.h ?

ddemidov, 2015-04-26:
I don't have any experience with servo motors, but the code in the demo does not specify an output port to use. So its happy with the first servo motor it sees. Did you try to specify an output port,  as in
```cpp
m1 = servo_motor(OUTPUT_A);
m2 = servo_motor(OUTPUT_B);
```
?

cho934, 2015-04-26:
I use January version of ev3dev.
In fact, servo_motor() don't take OUTPUT_X, but the id (0 to 7) of the servo, but it doesn't work.

dlech, 2015-04-26:
The argument for `servo_motor` is the port name. You can find the port name by reading the `port_name` attribute. It should be something like `in1:mux1`.

cho934, 2015-04-27:
It's working ! with ```in[ev3port_name 1-4]:sv[servo ID 1-8]```
Example :
```C++
servo_motor m_;
m_ = servo_motor("in3:sv1");
m_.set_position(50);
```

closed 2015-04-28 by cho934