ev3dev/ev3dev-lang #71: Not able to get motors or sensors working with examples

sharived, 2015-05-07:
I tried writing my own app and when in device::connect I get an exception which causes _path.clear() to execute and therefore think the motors are not connected.  Not sure why this is.  I tried the drive-test and ev3dev-lang-test apps as well and the same problem.  I upgraded to the latest version v0.6 of ev3dev with no change.  I checked my sys/class/tacho-motor files and they are there.  Do you know why this would happen?

sharived, 2015-05-07:
Note this is for cpp.

WasabiFan, 2015-05-07:
Is this using the source for the bindings from this repo's `develop` branch, or did you obtain the library from elsewhere (master branch)? Bindings that are compatible with recent motor drivers haven't been released yet (we hope to do it this week) so you'll need to be using them from the `develop` branch until then.

sharived, 2015-05-07:
I was using the master branch.  I'll give the develop branch a try.  Thanks!

WasabiFan, 2015-05-07:
Sorry for the confusion on that!

sharived, 2015-05-08:
I take that back, I was using the develop branch - didn't realize it was the default.  Should I wait a bit till you guys are done?  Or is there something I can do now to keep working?

WasabiFan, 2015-05-08:
If you were using the develop branch and it still wasn't working, we'll need to see what @ddemidov has to say on the matter -- I'm not familiar enough with the C++ binding to know what the issue is. But as far as I know, the C++ binding from this branch is functional.

ddemidov, 2015-05-08:
I am getting a compilation error in `ev3dev-lang-demo.cpp` (fixed in #72), but other than that, both `ev3dev-lang-test.cpp` and `drive-test.cpp` work fine for me.

When you say you upgraded to v0.6, what do you mean? I could not find an ev3dev repo tagged with this version. What kernel version do you use (the result of `uname -rv`)? The current one is 
```
3.16.7-ckt10-4-ev3dev-ev3 #1 PREEMPT Fri May 1 18:55:50 CDT 2015
```
and the C++ bindings should work with it. If this is the version you have, can you provide exact steps to reproduce the problem?

sharived, 2015-05-08:
I just did an update today.  I guess v0.6 was the brickman version - never mind.  uname -rv gives me
3.16.7-ckt4-ev3dev1 #2 PREEMPT Mon Jan 19 21:08:06 CST 2015

I have 2 EV3 Large motors plugged into ports B and C.  and the ultrasonic sensor in Port 1.

I am ssh'd into my EV3 over bluetooth.

When I run ev3dev-lang-test I get this

root@ev3dev0:/home# ./ev3dev-lang-test

No touch sensor found

No color sensor found

No ultrasonic sensor found

No gyro sensor found

No infrared sensor found

No i2c sensor found

No medium motor found

No large motor found

No dc motor found

No servo motor found

Brightness of left green led is 255

Trigger of right red led is [none]

Beeping...

Battery voltage is 7.38333 V

Battery current is 0.298666 A

and when I run drive-test I get this

root@ev3dev0:/home# ./drive-test

you need to connect an infrared sensor and large motors to ports B and C!

As mentioned I tried debugging it a bit and I seem to get into this code

In bool device::connect(const std::string &dir,...

        catch (...) { }

//REACHES here

        _path.clear();

So my path is always cleared.  I assume this is an exception that is being caught.  Any idea what might cause it?

ddemidov, 2015-05-08:
This is not the current kernel. You can not get the current one by upgrading, you need to reflash your SD card. See the announcement here: http://www.ev3dev.org/news/2015/05/01/Major-Release.

WasabiFan, 2015-05-08:
@sharived That version that you mentioned is not the newest -- but you can't upgrade any further as you would do it normally. With the most recent release, the motor drivers were completely overhauled, and these new drivers are the ones that we support. But due to some Linux disk magic that I admit I don't fully understand, existing installations of ev3dev can't be upgraded through `apt-get` as usual. Instead, you must reflash your SD card (see [here](http://www.ev3dev.org/news/2015/05/01/Major-Release/) for more details). After you reflash with the newest kernel version, you should be able to use our bindings.

P.S. This should be a one-time thing, so after this one upgrade you shouldn't be forced to start from scratch again.

WasabiFan, 2015-05-08:
@ddemidov Oh, looks like we commented at the same time ;)

sharived, 2015-05-08:
Thanks!  I'll try tomorrow.

sharived, 2015-05-09:
Finally got my SD card built and my C app works!  Thanks.  Now to get Python working again....

closed 2015-05-09 by sharived