ev3dev/ev3dev-lang #73: Equivalent of motor::running from earlier versions

dsharlet, 2015-05-08:
I've finally decided to follow through releasing the code for the project I've been working on with ev3dev, but it's now pretty seriously out of date. I've updated most of my code, but I'm stumped on something really simple: How do you tell if a motor is running or not? I see there is `stop_command` to get the stop command, but I don't see the equivalent of `command`, if that is even the right way to do this.

ddemidov, 2015-05-08:
There is a `state` attribute, which returns set of strings representing current motor state. So `motor.state().count("running")` should works as a test.

ddemidov, 2015-05-08:
Also see the new tutorial here: http://www.ev3dev.org/docs/tutorials/tacho-motors/

dsharlet, 2015-05-08:
Thanks, those docs are exactly what I needed.

closed 2015-05-08 by dsharlet

dlech, 2015-05-08:
@dsharlet, I had your ball catching robot in mind when I added the `run-direct` command. I'd be interested to know if you are able to make it work with other commands or if you have to use the `run-direct` command with your own PID as you did before.

dsharlet, 2015-05-08:
Thanks for that, I was pretty happy to see `run-direct` in the docs! I will also try to rebuild my motor class without using my own control loop and see how far I get and I'll let you know if I can make it work. It would be great to get rid of my custom servo class.

dsharlet, 2015-05-10:
@dlech, I've spent some time now updating my code. Everything seems to be working again, though I'm still using a custom controller with `run_direct`. I haven't been able to meet my requirements with the stock controller/API.

Here's a video that demonstrates what I'm trying to do (track a position with minimal delay): https://www.youtube.com/watch?v=vxx3bBVWaxg The video is old, but the behavior is the same today.

Here's the program used to generate that video (updated to latest ev3dev): https://github.com/dsharlet/ev3cv/blob/master/src/test/servo.cpp

I tried moving the call to "run_to_abs_pos" to outside the control loop, but the motor just stops as soon as it reaches the position setpoint and does not run again. I don't see an obvious candidate for another run command to use instead.