ev3dev/ev3dev-lang #77: Stopping/stop commands usability (C++)

dsharlet, 2015-05-10:
I have a minor suggestion regarding the stop commands, at least in the C++ API. I'm finding the API to be a bit confusing, `set_stop_command` feels a lot like a command to actually stop. However, it seems that `set_stop_command` is actually more analogous to the old `set_stop_mode` function, while the stop command actually stops. This might just be a matter of changing the documentation to describe this a bit more.

ddemidov, 2015-05-10:
I think this is not just C++. The bindings are using uniform specification, so any changes in the attribute names should be done by @dlech in [ev3dev/lego-linux-drivers](https://github.com/ev3dev/lego-linux-drivers) first.

Also, I think the docs at http://www.ev3dev.org/docs/drivers/tacho-motor-class/ are quite clear on this subject. Do you have any specific changes in mind?

ddemidov, 2015-05-10:
@WasabiFan, on a slightly related note, I am thinking on manually copying documentation strings from comments in [ev3dev/lego-linux-drivers](https://github.com/ev3dev/lego-linux-drivers) into `spec.json` structure. I would then use those to generate comments for C++ bindings and docstrings for python. Would you mind against this?

WasabiFan, 2015-05-10:
>  I am thinking of manually copying documentation strings from comments in ev3dev/lego-linux-drivers into  spec.json  structure. I would then use those to generate comments for C++ bindings and docstrings for python. Would you mind against this?

Nope; this is exactly what I was hoping for in adding the "description" fields in some of the objects. You should just be able to carry that throughout the spec.

dlech, 2015-05-10:
I'm open to changing this attribute to `stop_action` or something like that. Just open an issue in the main ev3dev issues if this is something you think we should do.

dsharlet, 2015-05-12:
I see now, I thought http://www.ev3dev.org/docs/tutorials/tacho-motors/ was the documentation, which is not quite as clear. I didn't realize there was a separate page of non-tutorial documentation...

closed 2015-05-12 by dsharlet