ev3dev/ev3dev-lang #88: Tacho motor doesn't work in position mode

yaqwsx, 2015-07-04:
Hello,
I tried to used the tacho motor in position (or hold) mode with the newest system image (20-5-2015). I set `position_sp` to 2000, and then run the `run-to-pos-abs` command. The motor didn't move. So I tried to set the PID constants. It is impossible to write or read from files hold_pid/k*. All operations fails with "Operation not supported". I think it is a bug. Could you look at it, please?

dlech, 2015-07-04:
You must also set the `duty_cycle_sp` if `speed_regulation` is `off` or set the `speed_sp` if `speed_regulation_ is on. The motor probably was "running", but had zero power.

`hold_pid/k*` are not implemented (yet) for this motor driver, so "Operation not supported" is the expected error message.

Also, these issues don't seem to have anything to do with the ev3dev-lang libraries, so the correct place to file this type of issue is at <https://github.com/ev3dev/ev3dev/issues>.

yaqwsx, 2015-07-04:
Thanks a lot! I wasn't able to find this information in documentation of the library. It's too bad that the constants aren't implemented yet, since the accuracy of the position regulator is quite terrible...

dlech, 2015-07-04:
> It's too bad that the constants aren't implemented yet, since the accuracy of the position regulator is quite terrible...

Pull requests are welcome!

WasabiFan, 2015-11-29:
Was this resolved/can it be closed?

closed 2015-12-18 by WasabiFan