ev3dev/ev3dev-lang #90: Provide an event based API for EV3 buttons

ddemidov, 2015-07-05:
Similar to how it is implemented in the `remote_control` class.

ddemidov, 2015-07-10:
Done for [C++](https://github.com/ev3dev/ev3dev-lang/blob/69f45d0936003a9bf3764cf8b3be525fc89c3c66/cpp/ev3dev.h#L1305-L1312) and exposed in Python. For now each button should be handled individually:
```py
def say_hello(pressed): if pressed: print('hello')
button.enter.onclick( say_hello )
while True: button.enter.process()
```
What I don't like about this is the processing part. I think I would prefer a single `process()` function that would iterate over all buttons. What is stopping me from implementing this is mainly where to put it (as a static method in a button class, or as a standalone helper class?) and how to name it.

Of course, it may be easily done on the user side of the code as well.

ddemidov, 2015-07-17:
I've added a static `button::process_all()` method. It simply calls `button::process()` for each of the EV3 buttons.

closed 2015-07-17 by ddemidov

cho934, 2016-03-25:
> Done for C++ and exposed in Python. For now each button should be handled individually:
```
def say_hello(pressed): if pressed: print('hello')
button.enter.onclick( say_hello )
while True: button.enter.process()
```
 I don't understand how to use this onclick and process function.
Could you create a small example using the onclick function in C++ ?

ddemidov, 2016-03-25:
Something like this should work:
```cpp
// Set a functor (here created as a lambda function) as the button event handler:
ev3dev::button::enter.onclick = [](bool state) {
    std::cout << state ? "pressed" : "not pressed" << std::endl;
};

// Do event processing in a loop. Whenever the enter button changes state,
// the onclick functor is called with the appropriate argument:
while (!ev3dev::button::back.pressed())
  ev3dev::button::enter.process();
```