ev3dev/ev3dev-lang #96: Better system for easily committing generated code in submodules?

WasabiFan, 2015-08-16:
Currently, If I run the autogen script in a clone of the repo, the changes are made to the submodule clone inside the main repo. If I then commit it, pushing it gets complicated because I need to manually specify the branch to pushed from a detached HEAD. If It then turns out that there are remote changes that I haven't pulled yet, it gets even more complicated.

Is there something that we can do to improve this? As far as I know, the submodules are our only option if we want to keep the core code in separate repos. So, do we want to add an option on the autogen script to specify the repo root to use (so that you could point it to a normal clone)? Or is there a better option that I'm not thinking of?

ddemidov, 2015-08-16:
I am not sure I understand the problem. Why does the HEAD have to be in a detached state? It probably is, initially, when you clone the super repo with all of its submodules, but then you can enter a submodule and checkout the master branch (or whatever you use as your main developement branch).

WasabiFan, 2015-08-16:
That's a good point; I was imagining that if I did that I'd be changing the ref that the submodule referenced to `master`, which would cause issues. But now that I think about it, you're probably right; I can checkout master without changing anything.

closed 2015-08-16 by WasabiFan