ev3dev/ev3dev-lang #101: Add `values` to the spec for specific sensor types

rhempel, 2015-09-13:
Unless I'm missing something, there's no easy way to get the `valuen` attribute for the `sensor` class.

What I would propose for now is that the `sensor` class gets 8 additional attributes, one for each `valuen` where n runs from 0 to 7

Eventually, it would be neat if we could have a mapping from the value name to n - depending on the mode.

ddemidov, 2015-09-13:
`value` is function taking integer argument - the number of the value to get. Is it not enough?

rhempel, 2015-09-13:
Yes, it's enough - I was just wondering if we could add to the spec attribute names that resolve back to the `valuen` attributes. 
For example, when we have the IR sensor in REMOTE mode, Channel 1 results are in the `value0` attribute.

ddemidov, 2015-09-28:
Sorry for being silent on this. So what you propose is having both `value(n)` function and `value0`...`value8` attributes?

rhempel, 2015-09-28:
No, I think I'm ok with the change I made to the `value()` function that accepts the index of the value you want. Maybe with Python/Lua or languages that have optional parameters, the default is to return an iterable (ie list) of all values. But  then the driver spec would need to be able to pass that information down to the common `value()` getter somehow.

rhempel, 2015-09-28:
Closing issue, the `value(n)` syntax is working fine - if we want to return an array of all values (which might be useful for I2C sensors that have a lot of values, then it will be a separate issue.

closed 2015-09-28 by rhempel