ev3dev/ev3dev-lang #103: Where I can find a Hello World example using Node.js for EV3Dev?

jabrena, 2015-09-16:
I would like to develop with Node.js some project with EV3Dev.
How to start?

ensonic, 2015-09-16:
connect via ssh to the brick and install nodejs from the debian repo first:

  apt-get install npm nodejs

Then follow https://github.com/WasabiFan/ev3dev-lang-js/blob/master/README.md

WasabiFan, 2015-09-16:
The instructions from @ensonic look good; just note that you should build from source (as described in the readme) because there isn't a recent enough version on npm. I plan to release a new version soon.

jabrena, 2015-09-16:
Ok, many thanks for the support.
I will try to follow the instructions.

Juan Antonio

jabrena, 2015-09-16:
I tried to read the API:
http://wasabifan.github.io/ev3dev-lang-js/

but some links are broken.
Is it the right API Docs for Node.js?

Juan Antonio

WasabiFan, 2015-09-16:
I know... I'm working on fixing that. The docs generator prefixes the page names with underscores, which GitHub interprets as special content and ignores. Sorry that the JS module is so disorganized at the moment! As a temporary measure, if you want to view the docs you can checkout the docs branch and serve it locally with your own http server.

dlech, 2015-09-16:
> apt-get install npm nodejs

Are these not distributed (installed) in the image already? They used to be and should be.

jabrena, 2015-09-17:
root@ev3dev:~/node# node helloWorld.js 
Hello World EV3

jabrena, 2015-09-17:
Hi, In what repo does exist examples to use the Node's API?

@WasabiFan Did you use some library to connect a webcam?

WasabiFan, 2015-09-18:
>  In what repo does exist examples to use the Node's API?

Are you talking about the ev3dev-lang-js API, or the generic Node API? If you're looking for examples on the ev3dev API, look [here](https://github.com/WasabiFan/ev3dev-lang-js/tree/master/examples). If you are looking for the generic Node.js environment, look at the [Node.js website](https://nodejs.org/en/).

> Did you use some library to connect a webcam?

I have not had any success with getting webcam images in Node, and that's not for a lack of trying :wink: If you figure out a way to do it, I'd love to hear about it!

jabrena, 2015-09-18:
Hi,

This weekend, I will test this packages:

https://www.npmjs.com/package/linuxcam
https://www.npmjs.com/package/v4l2camera

At the moment, I only have in my mind to detect colour blobs as NXTCam does.

Juan Antonio

jabrena, 2015-09-18:
Yes, I asked for examples using the API for EV3.
https://github.com/WasabiFan/ev3dev-lang-js/tree/master/examples

jabrena, 2015-09-19:
Hi,

I tried to upgrade node but latest version is a bit old:

``` shell

root@ev3dev:~# sudo apt-get install nodejs
Reading package lists... Done
Building dependency tree       
Reading state information... Done
nodejs is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@ev3dev:~# nodejs -v
v0.10.29

```

@WasabiFan, How to upgrade to node 0.12.7?

I think that next week node v4.0.0 (LTS) will be available to download.

jabrena, 2015-09-19:
@WasabiFan, Note: I could interact with a webcam, but I need to upgrade node version to use dependencies.

jabrena, 2015-09-19:
Maybe it is impossible to run latest node version with EV3.

```
root@ev3dev:~/node# cat /proc/cpuinfo 
processor	: 0
model name	: ARM926EJ-S rev 5 (v5l)
BogoMIPS	: 148.88
Features	: swp half thumb fastmult edsp java 
CPU implementer	: 0x41
CPU architecture: 5TEJ
CPU variant	: 0x0
CPU part	: 0x926
CPU revision	: 5

Hardware	: LEGO MINDSTORMS EV3 Programmable Brick
Revision	: 0000
Serial		: 0000000000000000

```

http://zo0ok.com/techfindings/archives/1820

ensonic, 2015-09-19:
+jabrena, yes, on the ev3 we're stuck with node-0.10. The v8 javascript engine dropped support for arm-v5 in 2012.

jabrena, 2015-09-19:
yes, but in this case, some libraries are not available for Node.js, isn't it?

ensonic, 2015-09-19:
@jabrena, that is correct. You will have to live with that unfortunately or choose a different language such as e.g. python. Right now I don't see that arm-v5 (and arm-v6) support is coming back to V8.

closed 2015-10-30 by jabrena