ev3dev/ev3dev-lang #105: Allow templates for a language binding to be in the language directory

rhempel, 2015-10-03:
Some of our language bindings exist in other repositories and are submodules in the `ev3dev-lang` repo.
Would it be a good idea to have the `template` folder for each language binding exist under the  folder the binding lives in?
I'll poke around in autogen to see if we can make this work without changing the command line too much

rhempel, 2015-10-03:
I can work around this easily enough like so:
```
#~autogen ../python/templates/python_generic-class classes.motor>currentClass
```

It makes assumptions about the relative locations of `autogen` and `python` but I think that's ON in this case. I started "fixing" autogen.js to handle looking in multiple locations, but then decided this was cleaner.

WasabiFan, 2015-10-04:
I like this idea. Can we reopen the issue? Personally, I'd rather take the time to make this work by default than use a workaround. I'll try to get this implemented soon.

rhempel, 2015-10-04:
Sure, I've reopened it. I was going to anyways because my "workaround" does not actually work. `autogen` silently says that the files are processed but nothing actually happens. The regexp in `autogen.js` needs to be fixed to allow more complex paths/names, possibly with quotes to allow arbitrary path/filenames.
On the other hand, I'd be happy to see what @WasabiFan comes up with!

WasabiFan, 2015-10-04:
@rhempel I just pushed some changes that add an entry into `autogen-list.json` that lets you specify the search directory for templates. If you change the one for python to point to a subfolder, it should resolve all your templates relative to that.

P.S. I noticed that the submodule that you currently have for the pure python binding is using an SSH clone URL; this should work on various Unices, but if one tries to clone it on Windows it may have unexpected results. I've been using HTTPS URLs for the other submodules.

rhempel, 2015-10-04:
Thanks, I keep forgetting to use HTTPS URLs when setting up submodules...
I'll try the new `autogen-list.json` soon.
When I was initially looking at working on `autogen` I was saving the language header and then adding that to the list of file to process, but I quickly realized that this was the wrong approach. Adding a `templateDir` is the right thing to do. Thanks!

rhempel, 2015-10-04:
Works great, but f you specify the template file, it needs to be relative to the autogen folder, like so:
```
        "templateDir": "../python/templates/"
```
@wasabifan, would you prefer this, or would you rather change `autogen.js`? In my opinion, the relative path is uglier, but it's less mysterious. It says, you can find the template for this binding up one level and in this path, relative to the folder `autogen` is running in.

WasabiFan, 2015-10-04:
That was an oversight on my part; I was intending for it to be relative to the `ev3dev-lang` root (so that the paths can just be `python/templates`, for example). I'll fix that now.

WasabiFan, 2015-10-04:
I just pushed changes to make it root-relative. Can you try it out and make sure that it worked?

rhempel, 2015-10-04:
Works great - even if you run it from somewhere else entirely. The paths are (I believe) relative to where `autogen.js` lives. Nice job! Marking closed again.

closed 2015-10-04 by rhempel