ev3dev/ev3dev-lang #107: autogen camel_case filter does not capitalize the first letter

rhempel, 2015-10-03:
In writing the `pure-python` bindings, I noticed that the camel-case filter does not capitalize the first letter - I worked around it like this:

```
{% assign class_name = currentClass.friendlyName | camel_case | capitalize %}
```

WasabiFan, 2015-10-03:
As I understand it, camelCase has a lower first letter (like a camel with its head down); I think you're thinking of PascalCase. I'm not sure if I have implemented a PascalCase filter, but the extra capitalize call should fix it (I can implement Pascal if you'd like).

dlech, 2015-10-03:
By definition, the first letter of camelCase is not capitalized. You want PascalCase.

https://en.wikipedia.org/wiki/CamelCase

dlech, 2015-10-03:
Beat me by three seconds :wink:

rhempel, 2015-10-03:
If you want to put in a PascalCase filter you can, the workaround works for me.

closed 2015-10-03 by rhempel