ev3dev/ev3dev-lang #110: Cannot run files owned by system users

TomHartogs, 2015-10-20:
I just created and compiled a C++ program writing "Hello World!" in the console. However when I try to launch the same programm from the brick itself I get an error message saying "Cannot run files owned by system users". I recall being able to run a Hello World program on the brick before. I already tried changing the permissions. Is this happening because I am working in the root folder?

ensonic, 2015-10-20:
Did you create a user accounton the ev3 already? For details see the 3nd comment at https://github.com/ev3dev/ev3dev/issues/393

Once you have a /home/robot, put your binaries there, make sure they are owned by the user (sudo chown -R robot: /home/robot) and then it should work.

dlech, 2015-10-20:
Since this issue doesn't have anything to do with the ev3dev-lang libraries, it should have been opened at https://github.com/ev3dev/ev3dev/issues instead.

The next version of brickstrap will not have this restriction, so this won't be a problem anymore.

dlech, 2015-10-30:
Moved issue to ev3dev/ev3dev#418

closed 2015-10-30 by dlech