ev3dev/ev3dev-lang #111: Does exist a Java example for EV3Dev?

jabrena, 2015-10-29:
Hi,

anyone has tested EV3Dev with Java?

Cheers

Juan Antonio

ddemidov, 2015-10-29:
Not that I know of. And its not listed on the [libraries](http://www.ev3dev.org/docs/libraries) page, so I guess it does not exist.

jabrena, 2015-10-29:
Yes, but i say an isolate example using the debian package.

``` java

public class HelloWorld {

    public static void main(String[] args) {
        // Prints "Hello, World" to the terminal window.
        System.out.println("Hello, World");
    }

}

```

ddemidov, 2015-10-29:
That should be possible as long as you install some implementation of java runtime.

jabrena, 2015-10-29:
I will try to test today.

Many thanks @ddemidov 

Cheers

jabrena, 2015-10-29:
Good night @ddemidov , I tested Java with EV3Dev without any problem.
This evening, I will try to define some issues to build a binding to Java as you do with Python.
https://github.com/jabrena/ev3dev-lang-java/

Cheers

jabrena, 2015-10-29:
https://github.com/jabrena/ev3dev-lang-java/issues/3

closed 2015-10-29 by jabrena