ev3dev/ev3dev-lang #113: Pillow instead of PIL for Python?

rhempel, 2015-10-29:
I don't know too much about image libs for Python but I have read that PIL is les actively developed lately and more devs are moving to Pillow...

Can anyone familiar with the topic chime in with an opinion please? Now's the time to decide on the standard image lib.

dlech, 2015-10-29:
I think you are already using Pillow

https://packages.debian.org/jessie/python-pil

rhempel, 2015-10-30:
Indeed, I am. I just did `apt-get install python-pil` on the EV3 over the Bluetooth link :-) Who needs WiFi? That port is much more useful as a high precision motion sensor/mouse.
So can I ask you to add that to the package list for the next release, and then close the issue?

dlech, 2015-10-30:
I assume we are packaging this (ev3dev-lang-python) as a debian package. As such, it should include python-pil as a dependency.

rhempel, 2015-10-30:
Agreed - so when I get around to building that Debian package I will close the defect.

ddemidov, 2016-01-06:
The debian package has bin built.

closed 2016-01-06 by ddemidov