ev3dev/ev3dev-lang #114: Java API for EV3Dev

jabrena, 2015-10-30:
Good morning,

I have created a repo to offer an API for Java developers using EV3Dev.
https://github.com/jabrena/ev3dev-lang-java

I think that Python development has good progress. For other language, what is the best strategy to integrate with the native API?
http://www.ev3dev.org/docs/sensors/
http://www.ev3dev.org/docs/motors/

I would like to create a easy example from Java to read a Sensor and use a motor for example.
Later, if the PoC goes well, I would like to generate the API for the rest of elements.
Is it possible to reuse low level development used from Python to other language?

Cheers

Juan Antonio

ddemidov, 2015-10-30:
You should look into [autogen](https://github.com/ev3dev/ev3dev-lang/tree/develop/autogen) system first. It has ev3dev specification in [spec.json](https://github.com/ev3dev/ev3dev-lang/blob/develop/autogen/spec.json), and it can generate code from the specification with liquid template files. You can look at the templates for python here: https://github.com/rhempel/ev3dev-lang-python/tree/master/templates.

If you make the system work, then (almost) the only thing you need to implement is base `Device` class (see python version [here](https://github.com/rhempel/ev3dev-lang-python/blob/91bfa71551d8a9f6649705fda68de904254ef745/ev3dev.py#L107-L191)) and the liquid templates. The rest of the classes with all the attributes are generated by autogen.

I think its completely OK to reuse Device implementation from either of the existing bindings.

jabrena, 2015-10-30:
Many thanks @ddemidov 

I will try to generate the infrastructure to run a basic example from Java and later, I start developing the whole support.

I think that working with the same technology, low-level problems could be solved in team.

Cheers

Juan Antonio

jabrena, 2015-10-31:
Hi @ddemidov, @dlech, @rhempel 

I have just tried to install dependencies with node, but I get the following error:

```js
➜  autogen  npm install
npm WARN package.json ev3dev-lang-autogen@1.0.0 No repository field.
npm WARN package.json ev3dev-lang-autogen@1.0.0 No license field.
npm ERR! git clone --template=/Users/jabrena/.npm/_git-remotes/_templates --mirror git://github.com/WasabiFan/liquid-node.git /Users/jabrena/.npm/_git-remotes/git-github-com-WasabiFan-liquid-node-git-344bec5b: xcrun: error: invalid active developer path (/Library/Developer/CommandLineTools), missing xcrun at: /Library/Developer/CommandLineTools/usr/bin/xcrun
npm ERR! Darwin 15.0.0
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install"
npm ERR! node v4.0.0
npm ERR! npm  v2.11.2
npm ERR! code 1

npm ERR! Command failed: git clone --template=/Users/jabrena/.npm/_git-remotes/_templates --mirror git://github.com/WasabiFan/liquid-node.git /Users/jabrena/.npm/_git-remotes/git-github-com-WasabiFan-liquid-node-git-344bec5b
npm ERR! xcrun: error: invalid active developer path (/Library/Developer/CommandLineTools), missing xcrun at: /Library/Developer/CommandLineTools/usr/bin/xcrun
npm ERR! 
npm ERR! 
```

It says something about a problem an invalid active developer path.

Any idea to fix the problem?

Cheers

Juan Antonio

dlech, 2015-10-31:
@WasabiFan is the node.js guy.

It looks like maybe you are trying to install this on a Mac. It is meant to install on the EV3 itself.

ddemidov, 2015-10-31:
you need to run `npm install` from `autogen` folder. If that does not work, we need to ask @WasabiFan, he is the author of autogen system.

ddemidov, 2015-10-31:
And it will work out of ev3, no need to install it there.

dlech, 2015-10-31:
Oh, wait. You are tying to run the autogen. I though you were trying to run the node.js binding. So never mind.

dlech, 2015-10-31:
Do you have XCode command line tools installed?

jabrena, 2015-10-31:
I think that I don't have that tool:

```bash
➜  autogen  XCode
zsh: command not found: XCode
```

Is it necessary to be installed XCode on Mac? 
Is it enough with gcc?

jabrena, 2015-10-31:
I read a bit and I have installed them:

```bash
➜  autogen  xcode-select -p
/Library/Developer/CommandLineTools
```

jabrena, 2015-10-31:
I was reading a bit.

I am installing

```bash
xcode-select --install
````

I will test again, in a while

jabrena, 2015-10-31:
Installed.

```bash
➜  autogen git:(develop) node autogen.js
Queuing 5 file groups...
Completed processing file "python/spec_version.py"
Completed processing file "cpp/ev3dev-lang-demo.cpp"
Completed processing file "js/sensor.ts"
Completed processing file "cpp/ev3dev.cpp"
Completed processing file "js/motor.ts"
Completed processing file "js/extras.ts"
Completed processing file "cpp/ev3dev-lang-test.cpp"
Completed processing file "cpp/ev3dev.h"
Completed processing file "wrapper-specification.md"
Completed processing file "python/ev3dev.py"
Completed processing file "lua/ev3dev.lua"
```

I suppose that I have observe the way for the rest of language to to the same for Java, isn't it?

dlech, 2015-10-31:
Yes. Python has most recently been updated, so use that for your guide.

jabrena, 2015-10-31:
Ok, Tomorrow, I will research a bit about the system.
Many thanks

WasabiFan, 2015-11-01:
Looks like this was resolved. Is there anything I should change about the autogen system to make it clearer/work better?

P.S. Sorry I haven't been involved in the discussions these past weeks -- very busy! I plan to try to get caught up soon.

jabrena, 2015-11-01:
Hi,

the process to install the autogen infrastructure is clear, maybe we could update the documentation to add the commands for Mac to ensure that everybody with mac has Xcode installed.

```
xcode-select --install
```

Once, the developer has installed the system, the developer test autogen with the node command:

```
node autogen.js
```

for python, the parts generated by the templates defined on:
https://github.com/rhempel/ev3dev-lang-python/tree/master/templates

``` bash
Completed processing file "python/spec_version.py"
Completed processing file "python/ev3dev.py"
```

the system binds with templates in python in the file:

autogen-list.json

```json
    "python": {
        "files": [
            "python/ev3dev.py",
            "python/spec_version.py"
        ],
        "templateDir": "python/templates/"
    },
```

but what is the mechanics?

I have added a local copy of the repo for java:
https://github.com/jabrena/ev3dev-lang-java

with 2 files:

* ev3dev.java
* spec_version.java

in both cases, I have used python idea.

In autogen folder i updated the file to add java languaje:

```json
    "java": {
        "files": [
            "java/ev3dev.java"
        ],
        "templateDir": "java/templates/"
    },
```

ev3dev.java:

//~autogen ../java/templates/java_spec_version
spec_version = "spec: 0.9.3-pre-r1, kernel: v3.16.7-ckt10-4-ev3dev-ev3"

//~autogen

I suppose that autogen get the content of .liquid template and past the content in the file, but currently with this simple idea, i receive the following error:

``` bash

➜  autogen git:(develop) ✗ node autogen
Queuing 6 file groups...
Completed processing file "python/spec_version.py"
/Users/jabrena/Documents/DATA/2015/RESEARCH/robotics/ev3dev/java/ev3dev-lang/autogen/autogen.js:189
    if (commentInfo.start instanceof RegExp)
                   ^

TypeError: Cannot read property 'start' of undefined
    at processNextAutogenBlock (/Users/jabrena/Documents/DATA/2015/RESEARCH/robotics/ev3dev/java/ev3dev-lang/autogen/autogen.js:189:20)
    at /Users/jabrena/Documents/DATA/2015/RESEARCH/robotics/ev3dev/java/ev3dev-lang/autogen/autogen.js:175:9
    at FSReqWrap.readFileAfterClose [as oncomplete] (fs.js:380:3)

```

If you add the in .gitmodules this part:

```bash
[submodule "java"]
	path = java
	url = https://github.com/jabrena/ev3dev-lang-java
	branch = master
````
you could observe the error.

Juan Antonio

ddemidov, 2015-11-01:
Here is what you need to do:

1. Add a line for java [here](https://github.com/ev3dev/ev3dev-lang/blob/10c1c7e18cf49c4ed4e9b9a44e0f92f45bdb0e63/autogen/autogen.js#L20). You may just copy the line for `cpp` I think. This will tell autogen how its markers look like in java files.
3. After this, autogen will look for lines starting with `//~autogen` in the files that you specifed [here](https://github.com/ev3dev/ev3dev-lang/blob/10c1c7e18cf49c4ed4e9b9a44e0f92f45bdb0e63/autogen/autogen-list.json). Start of the autogen block should look like this:
```
//~autogen java_spec_version
```
There is no need to specify full relative path as long as you've told autogen [where to look for the templates](https://github.com/ev3dev/ev3dev-lang/blob/10c1c7e18cf49c4ed4e9b9a44e0f92f45bdb0e63/autogen/autogen-list.json#L16). I know python `spec_version.py` does this, but this is a leftover overlooked by me and @rhempel.

By the way, you can run autogen just for java by `node ../autogen/autogen.js java`. This way you won't touch other people's bindings and won't commit the changes accidentally.

jabrena, 2015-11-01:
Right!!!

Now, I begin seeing the light!!

Updating the file:

``` js
    '.py': { start: /#~autogen *([\w-]+) *((\s*[\w\."']+>[\w\.]+)*)/, end: "#~autogen" },
    '.java': { start: cStyleAutogenStart, end: cStyleAutogenEnd }
```
I can update the files:

```  java
➜  autogen git:(develop) ✗ node autogen.js java
Queuing 1 file groups...
Completed processing file "java/ev3dev.java"
Completed processing file "java/spec_version.java"
```

Besides, you have reason, sending the parameter java is safer to not update other bindings.
This afternoon, I will continue researching more about templates. 

Juan Antonio

rhempel, 2015-11-01:
Welcome to the club Juan Antonio - I think you will like this project and we look forward to your contributions on the Java binding.

jabrena, 2015-11-01:
hehehe,

Yes this idea is Smart because the system deacoplates languages from low level and this is the way to increase the scalability of the project. Anyway, I will need some hours to get the first result for Java binding. 

I continue with the researching!!! :D

WasabiFan, 2015-12-19:
This is going to be held back until next release. Initial PR in #134.