ev3dev/ev3dev-lang #115: PiStorms platform devices

dlech, 2015-10-31:
This is what PiStorms is looking like now.

- [ ] Leds:

        pistorms:BA:red:ev3dev
        pistorms:BA:blue:ev3dev
        pistorms:BA:green:ev3dev
        pistorms:BB:red:ev3dev
        pistorms:BB:green:ev3dev
        pistorms:BB:blue:ev3dev 

    `pistorms:BA` and `pistorms:BB` are each a "single" LED, so the 3 colors can be mixed.

- [ ] Buttons:

        KEY_ENTER (the GO button)
        BTN_TOUCH (touchscreen)

    As mentioned previously, on PiStorms, the buttons don't have a fixed path, so the current implementation of buttons will probably have to change

- [ ] Power Supply:

        pistorms-battery (voltage only)

- [ ] Input Ports:

        pistorms:BA:S1
        pistorms:BA:S2
        pistorms:BB:S1
        pistorms:BB:S2

    I believe these should be picked up once I update the kernel documentation on ev3dev.github.io.

- [ ] Output Ports

        pistorms:BA:M1
        pistorms:BA:M2
        pistorms:BB:M1
        pistorms:BB:M2

`B` stands for "Bank", so you might want to translate `pistorms:BX` to `PistormsBankX` in the code gen.

jabrena, 2015-10-31:
It sounds nice! 
I hope to receive a unit in some weeks. I will help in the testing process.

ddemidov, 2015-10-31:
> B stands for "Bank", so you might want to translate pistorms:BX to PistormsBankX in the code gen.

I did not understand this part. Do you mean user-friendly name for the port should be `PistormsBankX`?

dlech, 2015-10-31:
Yes, that is what I mean. Just a suggestion.

ddemidov, 2015-11-01:
@dlech, we can do that, or we can stick to `INPUT_X`, `OUTPUT_X` on all platforms. The latter would make scripts more portable between platforms. What do you think?

dlech, 2015-11-01:
I think that sounds good. I would like the same program to run on any platform without having to make any changes.

WasabiFan, 2015-12-19:
What's the conclusion on this? Were the PiStorms devices added to Python (if so, what was done there)? If not, I think we need to have a discussion on the platform differences and standardization.

As I understand it, there are three platforms that we need to support as of now (correct me if I'm wrong):
- EV3 brick
- BrickPi
- PiStorms

So, I think we need to get this standardized spec-wise so that adding more platforms (I think there was talk of the Beaglebone) isn't going to break anything.

For the port names, @ddemidov How do you think we can implement the ability to use  the same `OUTBUT_X` constants? In my case at least, I declare them as `export var OUTPUT_A = 'outA'`, so they are used directly as strings. Did you already find a way to implement this on your end?