ev3dev/ev3dev-lang #116: Send commands to outA or outB

jabrena, 2015-11-01:
Hi,

I am using some motors connecting and disconnecting from motor port 1 and 2 and now, I have 2 NXT motors in motor port A y B. Reading values from sysfs:

root@ev3dev:~# ls /sys/class/tacho-motor/
motor2  motor4

I observe this answer. If I use this way:

root@ev3dev:~# for f in /sys/class/tacho-motor/*; do echo -n "$f: "; cat $f/port_name; done
/sys/class/tacho-motor/motor2: outB
/sys/class/tacho-motor/motor4: outA

I see an absolute reference, because motor2 is connected in port B and motor4 is connected in port A.

How to send commands to motor port A & B?

http://www.ev3dev.org/docs/tutorials/tacho-motors/

ddemidov, 2015-11-01:
I am not sure what is the problem here. Is this not what you would expect?

jabrena, 2015-11-01:
imagine that I write a program which the robot has 2 motor connected to motor port A & B. 

in the preliminar tests, I have connected 2 motors, but the reference is motor2 & 4 but they are connected to port A & B. 

I would like to send:

```bash
root@ev3dev:~# echo 50 > /sys/class/tacho-motor/outA/duty_cycle_sp
root@ev3dev:~# echo run-forever > /sys/class/tacho-motor/outA/command
root@ev3dev:~# echo 50 > /sys/class/tacho-motor/outB/duty_cycle_sp
root@ev3dev:~# echo run-forever > /sys/class/tacho-motor/outB/command
```

does exist some way to make reference for motor port based on the position where user connected?

ddemidov, 2015-11-01:
If you look at how Device [is initialized in python now](https://github.com/rhempel/ev3dev-lang-python/blob/5df2cadf292cf8ac7ee8af5998e0a41089f8b84a/ev3dev.py#L117-L160), it tries to connect to any device by default, but if user provides `port_name` (well, any attribute value), it tries to match that.

ddemidov, 2015-11-01:
So you just iterate over devices until you find the one you need.

jabrena, 2015-11-01:
Oki, I understand the idea.

When you init a Motor object, you search the element in "/sys/class/tacho-motor" and you fix the object with that element.

Thanks

closed 2015-11-01 by jabrena