ev3dev/ev3dev-lang #118: Updating the JS binding for recent changes

WasabiFan, 2015-11-11:
@ddemidov Can I get a list of the changes that I need to adapt to on my Node.js side? I believe that there were changes to the LED spec, but didn't follow closely enough to fully grasp the changes that I should make. Also, what's the situation regarding supporting the various ev3dev platforms (RasPi, BB, etc)? Do we need to change the default instances?

P.S. Sorry I've been so out-of-touch :wink:

ddemidov, 2015-11-11:
Here are the changes, I hope did not miss anything. Btw, thanks for opening this, it will also be useful for C++ bindings.

**Motors**

* Added [specification](https://github.com/ev3dev/ev3dev-lang/blob/3bfd016eb809a164dc1b0557aa639aa88d46fff4/autogen/spec.json#L346-L365) for `LargeMotor` and `MediumMotor`.

**Sensors**

* Added [specification](https://github.com/ev3dev/ev3dev-lang/blob/3bfd016eb809a164dc1b0557aa639aa88d46fff4/autogen/spec.json#L1080-L1090) for `TouchSensor`.

**Buttons, RemoteControl**

* We have created a `ButtonBase` class that provides some basic functionality for buttons and remote control. This is not in specification, but I think is worth noting.
* Added [mapping](https://github.com/ev3dev/ev3dev-lang/blob/3bfd016eb809a164dc1b0557aa639aa88d46fff4/autogen/spec.json#L996-L1023) between `InfraredSensor` values in `IR-REMOTE` mode and `RemoteControl` buttons.
* Added per platform [specification](https://github.com/ev3dev/ev3dev-lang/blob/3bfd016eb809a164dc1b0557aa639aa88d46fff4/autogen/spec.json#L346-L365) for GPIO files and button values for EV3 and BrickPI.

**Leds**

This is the biggest change in specification.

* Led interface was switched to group-based. We used to group EV3 leds by color. That is, we had `red` group of 'red_left' and `red_right` leds and `green` group of `green_left` and `green_right`. That only allowed us to operate on all leds at once to create colors like `amber` or `orange`. Group based interface makes led groups based on their location. We have `left` and `right` groups for EV3. This allows the user to think in terms or logical `led` that can create different colors. For example, the following code is now supported in python:
```py
# Set left logical led to red color.
Leds.set_color(Leds.LEFT, Leds.RED)
# Set trigger for the right logical led, make it 50% bright:
Leds.set(Leds.RIGHT, trigger='timer', brightness_pct=0.5)
```
This also scales well onto other platforms. For example, PiStorms have two logical groups, each having Red, Blue, and Green physical leds.
* Added per-platform [specification](https://github.com/ev3dev/ev3dev-lang/blob/3bfd016eb809a164dc1b0557aa639aa88d46fff4/autogen/spec.json#L1193-L1211) for led groups and colors.

**Sound**

* This is not in specification, but we changed how sound works in python. Instead of using platform-specific sysfs node, we switched to using `/usr/bin/beep` for making beeps. See description of sound interface [here](http://python-ev3dev.readthedocs.org/en/latest/?badge=latest#ev3dev.ev3.Sound).

The way we currently work with different platforms may change in the future, see discussion in ev3dev/ev3dev#412. Also see #115.

WasabiFan, 2015-11-11:
Great! Thanks. I'll start working on these changes today.

WasabiFan, 2015-11-11:
@ddemidov I noticed that the `systemDeviceNameConvention` uses a `{0}` to denote the numbers in a name. I think that was something I added as a test a while ago, and now we both replace that with a `*` for use. Should we just make that replacement in the spec?

WasabiFan, 2015-11-12:
Oh, and what's the current status on detecting the running platform for buttons/LEDs? I saw a long discussion somewhere referencing a fair amount of Linux magic that I didn't understand... was that concluded? If not, what are you doing for C++ and Python in this area?

ddemidov, 2015-11-12:
> and now we both replace that with a * for use. Should we just make that replacement in the spec?

I don't see why not.

> Oh, and what's the current status on detecting the running platform for buttons/LEDs?

I think that is still in the planning phase. I made rhempel/ev3dev-lang-python#90 with this in mind, but nothing besides that.