ev3dev/ev3dev-lang #123: c# bindings

pgrudzien12, 2015-11-22:
I've written fullly autogenerated C# binding based on this project. Do you accept such PR's or the list of languages are fixed/frozen? (I've seen #114 so I suppose not but..)

I'm still not sure if you would need anything more than autogeneration to make it into this repo.

rhempel, 2015-11-23:
No, the list of languages is not frozen - I have one person that would be very happy indeed to see a C# binding. Please feel free to issue a PR!

WasabiFan, 2015-11-23:
I'd love to have a C# binding in this repo! Personally, C# is my favorite programming language; I've been thinking about trying to get one started for a while, but it hadn't seemed like they had gotten far enough to make the CoreCLR run on ARM Linux, and I had had trouble with mono on the EV3.

How are you deploying the code to the EV3? Did you configure a build job from Visual Studio, or are you just copying the files manually?

> I'm still not sure if you would need anything more than autogeneration to make it into this repo.

I wouldn't say that autogen is a requirement so much as it is recommended to make your life easier. If you think you're fine maintaining the code manually, go for it. Mainly, we're just looking for high-quality wrappers that support at least the core functionality of ev3dev. We also generally want the interface to look like the one described in [our spec](https://github.com/ev3dev/ev3dev-lang/blob/develop/wrapper-specification.md) to keep things consistent, but it doesn't need to be exact if there are language features that you'd like to take advantage of or you just think there's a better way to do it.

PRs are welcome; if you think you have a binding that is reasonably functional, open a PR to add a submodule and we'll take a look.

P.S. I knew I recognized your username... you're the one who got ASP.NET 5 running on the EV3 (and then [was featured](https://twitter.com/shanselman/status/667144339353526272) by Scott Hanselman on Twitter)! That was pretty cool.

I have some friends that work on the ASP.NET team at Microsoft; a lot of them know about your project :wink:

dlech, 2015-11-23:
Unfortunately, the "arm" in CoreCLR is "armhf" in Debian, so no CoreCLR for the EV3 (which is "armel" in Debian). The latest mono from Xamarin works though.

WasabiFan, 2015-11-23:
> Unfortunately, the "arm" in CoreCLR is "armhf" in Debian, so no CoreCLR for the EV3 (which is "armel" in Debian). The latest mono from Xamarin works though.

Yeah, that's the conclusion that I had come to... no official support :disappointed: But at least Xamarin is continuing to maintain their version, and the debug tools in Visual Studio _should_ work with mono.

rhempel, 2015-11-23:
Wow - that Microsoft connection is pretty cool!

pgrudzien12, 2015-11-23:
@WasabiFan Hey I'm happy I can help :) I've just created PR to add C# bindings to this repo. I was also struggling with mono on ev3 but fortunately I documented installation on blog you mentioned. And it's great to hear that this project had some attention. I think you will benefit from this as well - it was based mainly on ev3dev !

This code generator I PR'ed should be fully compatible with your latest API but I haven't tested it extensively yet. 

I think it still lacks a few important pieces so adding a few things wouldn't hurt (if you don't mind). I think it would be beneficial to see things like:

- nuget spec
- tests based on files as discussed in #112
- maybe demo
- md file to explain or document process

Thank you very much for your work guys! Without you running DNX on Lego would be impossible.

pgrudzien12, 2015-11-23:
I think I was too eager to make this PR :) I'll work on it a little more and then try it again.

pgrudzien12, 2015-11-29:
I created another PR #129

tim241, 2016-11-29:
Because this project looks dead, I tried to follow the instructions from @pgrudzien12 on his blog( http://bleedingnedge.com/2015/11/08/asp-net-5-on-lego-mindstorms-ev3-using-ev3dev/ ) and it didn't work out on the newest release, so I am currently trying to get mono to compile on the ev3 (because mono didn't want to install) but the batteries keep dying, so after mono is compiled I'll try to write a simple dll that will work with the ev3dev to control the motors and sensors. the compiling is very slow on the ev3dev though, it takes more than a day to compile and I don't even know if it'll fully compile, I really hope I can make a C# binding for the ev3dev because that will be cool :+1:

WasabiFan, 2016-11-29:
If you do, let us know! I imagine you can leverage a fair amount of the code that's already here, excluding the driver interface changes that have been made since this code was written. I have historically gotten mono running on the brick I believe, but never used it for anything other than a quick test; can confirm, it's really slow to get installed!

bmegli, 2016-11-29:
@tim241,

I haven't used bindings, but I used Mono and C# on EV3 with some instructions [here](https://github.com/bmegli/xv11lidar-test/tree/master/examples/csharp/xv11pipe2csv).

Maybe it can be of some help.

I could even compile with Visual Studio on PC and run with Mono on EV3. 

The build process there was through makefiles, for both Mono `msc` and VS `csc`.  Makefiles are probably not the most handy build system for for C# people but maybe it can be of some help.

bmegli, 2016-11-29:
@tim241 

> but the batteries keep dying

I sometimes leave my EV3 powered on for whole day with charger plugged to [EV3-Rechargable-Battery](https://shop.lego.com/en-US/EV3-Rechargeable-DC-Battery-45501).

It kind of changes the way you work when you don't have to worry about batteries and don't have to rebuild robot for replacing batteries/charging.

tim241, 2016-11-29:
sorry @bmegli but €109,99 is too much money to spend for me...I'm only 15 years old and get 5,50 /week so I can't afford that

dlech, 2016-11-29:
Something like [this](https://www.adafruit.com/products/63) and [this](https://www.adafruit.com/products/1328) may be more in your price range if you need to leave the EV3 on for long periods of time.

Also, I would suggest using docker to install Mono and create an image with brickstrap. For example you can use the following `Dockerfile`

```
FROM ev3dev/ev3dev-jessie-ev3-generic

RUN apt-get update --yes && apt-get install --yes mono-complete
```

`mono-complete` is probably a bit excessive but then you won't be missing anything. :smiley:

dlech, 2016-11-29:
`mono-complete` is getting hung up for me. It installs a bunch of desktop packages that we don't need on EV3 anyway. I did successfully make an image that has `mono-runtime` installed though. You can download it [here](https://oss.jfrog.org/artifactory/oss-snapshot-local/org/ev3dev/brickstrap/2016-11-28/snapshot-ev3dev-jessie-ev3-mono.img.xz).

tim241, 2016-11-29:
@dlech I tried installing ```mono-runtime``` or even ```mono-complete``` but I got errors while installing the assemblies, I'll try the docker thing tomorrow also thanks :+1:

tim241, 2016-11-29:
@dlech Thank you :+1: I also have errors while installing mono-complete but I thought compiling it on the brick itself should work better until I discovered that it would take days to compile and due the problems with the batteries(will be fixed thanks to you) I was stuck...thank you :+1:

tim241, 2016-11-29:
@dlech so the upcoming days I will be trying to program a dll the ev3..that can only be executed on the ev3 , thank you very much :+1: :)

WasabiFan, 2016-11-29:
@tim241 The image that @dlech linked to should have that package preinstalled if you don't want to wait.

tim241, 2016-11-29:
@WasabiFan I know..I'll test tomorrow because I am busy now :+1:

tim241, 2017-05-21:
Sorry, was working and doing other things, but I'm back to this again :+1: