ev3dev/ev3dev-lang #128: Notes on possible functionality improvements to autogen

WasabiFan, 2015-11-26:
Here's a quick list of features I think it might be worth adding (for a TODO list / general discussion if needed):
- [ ] Ability to mark sections of code to be templated inline for sections that aren't worth a new template file. This would allow you to do something like this (although probably with slightly different syntax):
```js
doAThing();
//~autogen
doOtherImportantThing({{myClass.myProp}});
//~autogen
```
- [ ] Ability to include templates in other templates (mark a template to be re-evaluated after initially being inserted)

ddemidov, 2015-11-26:
> Ability to mark sections of code to be templated inline

What will happen to this after the first generator run? The liquid code will be replaced by normal code, right? So the code has no special meaning to autogen system, and subsequent generator runs won't be able to update the code?

WasabiFan, 2015-11-26:
That's my "slightly different syntax" note :wink: I'm still thinking about it -- do you have any ideas there? My hope was that it would be autocompleted and syntax highlighted by an IDE normally, so it couldn't just be in a comment... hmmmm :confused:

ddemidov, 2015-11-26:
I have no ideas. I don't see how we can make same code to be accepted both by compiler/interpreter and autogen unless its in a comment.