ev3dev/ev3dev-lang #130: [Doubt] I can't turn a determinated number of degrees.

jabrena, 2015-12-08:
Good morning,

I was using a Regulated motor: ( A NXT Tacho Motor)
http://www.philohome.com/nxtmotor/nxtmotor.htm

to test how to learn to write a method to rotate and rotateTo but bot ways described in the docs doesn't work:
http://www.ev3dev.org/docs/tutorials/tacho-motors/

rotateTo:

``` bash
echo 50 > duty_cycle_sp
cat position_sp
180 <- output
echo 270 > position_sp
cat position_sp
270
echo run-to-abs-pos > command
```

If I execute this set of commands, I can't observe a motor response. What is my problem?

I have the same problem with a relative rotation:

``` bash
echo 50 > duty_cycle_sp
cat position_sp
180 <- output
echo 270 > position_sp
cat position_sp
270
echo run-to-rel-pos > command
```

Any help?

I am using last version of EV3Dev:

``` bash
root@ev3dev:/sys/class/tacho-motor/motor1# uname -a
Linux ev3dev 3.16.7-ckt19-8-ev3dev-ev3 #1 PREEMPT Mon Nov 9 11:03:41 CST 2015 armv5tejl GNU/Linux
```

ddemidov, 2015-12-08:
Is speed regulation enabled? (`cat speed_regulation`). If yes, you need to use `speed_sp` instead of `duty_cycle_sp`.

jabrena, 2015-12-08:
Hi @ddemidov, 

Yes, the motor was not enabled the speed regulation.

``` bash
root@ev3dev:/sys/class/tacho-motor/motor0# cat speed_regulation
off
```

So, I enabled:

``` bash
echo on > speed_regulation
```
and the code run nice.

What is the difference between speed_sp and duty_cycle_sp?

Why a tacho-motor has speed regulated disabled? I think that motors should have the speed regulation by default, isn't it?

I think that it is the main difference between a DC-Motor, isn't it?
What is your opinion?

Juan Antonio

ddemidov, 2015-12-08:
> the motor was not enabled the speed regulation.

I am confused then, because in that case your initial code with `duty_cycle_sp` should work initially.

> What is the difference between speed_sp and duty_cycle_sp?

Here is a quote from http://www.ev3dev.org/docs/drivers/tacho-motor-class/:

> speed_regulation (read/write)
Turns speed regulation on or off. If speed regulation is on, the motor controller will vary the power supplied to the motor to try to maintain the speed specified in speed_sp. If speed regulation is off, the controller will use the power specified in duty_cycle_sp. Valid values are on and off.

So `duty_cycle_sp` controls power applied to the motor (0 to 100%) and `speed_sp` controls rotational speed of the motor. When motor is under load, then with speed regulation off it will rotate slower. When speed regulation is enabled, driver will apply more power in attempt to keep the rotational speed constant.

jabrena, 2015-12-08:
I am going to reboot EV3Dev in order to reset motor values stored in memory.
I am going to do a test now and I will inform.

Many thanks for the info about the attributes: duty_cycle_sp & speed_sp

the range of values for duty_cycle_sp is 0-100
what is the range for speed_sp?

ddemidov, 2015-12-08:
> what is the range for speed_sp?

http://www.ev3dev.org/docs/tutorials/tacho-motors/ says

>  the units are tachometer counts per second. You can convert to RPM by dividing the value in count_per_rot. For the EV3 large motor, the maximum speed is about 900 counts per second and the EV3 medium motor is about 1200 counts per second. To ensure that the motor runs at the same speed every time, event with low battery, use values less than these.

ddemidov, 2015-12-08:
> I am going to reboot EV3Dev in order to reset motor values stored in memory.

You could also `echo reset > command`.

WasabiFan, 2015-12-08:
@jabrena Remember that we maintain the docs on the website for the purpose of learning about ev3dev interfaces. When you discover that something isn't working as you would expect it to, open up <www.ev3dev.org>, search for the device in question ("motor" in this case), and read through the information on the related pages. Many queries such as this one are answered there. You can also check the tutorials section to see if there's anything there.

If you find information that isn't in the docs, or isn't easily accessible, that's when you should make sure to open an issue to help others in the future find the information. And when you open issues about core ev3dev (not the language libraries in ev3ev-lang), it's best to open them at the ev3dev meta repo to make sure that the right people see it and that future visitors can find it.

jabrena, 2015-12-08:
Hi mates @ddemidov @WasabiFan ,

I will follow the rules. Besides, if you want, I could help in docs too. I think that my questions are basic, but I don't have strong experience with the API.

WasabiFan, 2015-12-08:
> I will follow the rules. Besides, if you want, I could help in docs too.

To be clear, I'm not trying to discourage you from participating in discussions, filing bugs, etc., nor am I trying to enforce strict rules; I am, however, attempting to make sure that we keep our information organized. We have 30+ repos in the broader organization with at least 3 separate issue trackers, as well as our online docs and (now-defunct) wiki -- this is a lot to keep track of both for maintainers and for new users. We need to make sure that the information that people need is where they can find it.

We (both ev3dev-land specifically and the broader ev3dev) welcome contributions of all sorts -- If you believe that this information wasn't easy enough to find, open an issue about it on the main repo to discuss it. Or, better yet, open a PR for our site with a proposed fix :wink:

jabrena, 2015-12-08:
@WasabiFan you explaining in good way. (I was writing the new issue)

closed 2015-12-19 by WasabiFan