ev3dev/ev3dev-lang #156: How to use with the autogen system?

mob41, 2016-03-31:
I just finished almost 3/4 of the unified library. And started to use autogen system. I just checked with other unified code's autogen setup on their code.

Is it the some of the code will be automatically generated inside the new line?
```java
//~autogen generic-classes classes.motor>currentClass
               //A new line
//~autogen
```

Besides, there is something like this:
```java
//-----------------------------------------------------------------------------
```
in the code. Is it necessary to put it inside the code, to separate the code?

I just finished one of the classes (Motor) with the autogen system.
[https://github.com/mob41/ev3dev-lang-java/blob/master/src/main/java/ev3dev/hardware/motors/Motor.java](https://github.com/mob41/ev3dev-lang-java/blob/master/src/main/java/ev3dev/hardware/motors/Motor.java)
I would like to check that is it correct to prevent doing it with the same problem on the other class.

WasabiFan, 2016-03-31:
I don't have time to check it/explain now, but I will be able to write up an explanation later today if someone else doesn't beat me to it.

ddemidov, 2016-03-31:
Have a look at http://ev3dev-lang.readthedocs.org/en/latest/autogen.html for a short explanation. Please tell us if the explanation is not clear enough or needs anything more.

Basically, you need a `templates` folder in your repo, where you put liquid templates that are used to generate code using data from [spec.json](https://github.com/ev3dev/ev3dev-lang/blob/develop/spec.json).

For example, look at the [rhempel/ev3dev-lang-python/templates/motor_commands.liquid](https://github.com/rhempel/ev3dev-lang-python/blob/master/templates/motor_commands.liquid) template. It is used to generate [this section](https://github.com/rhempel/ev3dev-lang-python/blob/master/ev3dev/core.py#L680-L741) of the core.py file. Comment in [line 680](https://github.com/rhempel/ev3dev-lang-python/blob/master/ev3dev/core.py#L680) tells the autogen system to use `motor_commands` template and to pass [classes.motor](https://github.com/ev3dev/ev3dev-lang/blob/develop/spec.json#L7) struct as the `currentClass` parameter. The template is written using liquid syntax described here: https://shopify.github.io/liquid/basics/introduction.

You also need to create an `autogen-config.json` file in the top directory of your repo. It lists files that should be visited by the autogen script. Look at the [python's version](https://github.com/rhempel/ev3dev-lang-python/blob/master/autogen-config.json).

Finally, you should be able to run the autogen script with `./ev3dev-lang/exec-autogen.js`.

ddemidov, 2016-03-31:
> Is it the some of the code will be automatically generated inside the new line?

Yes, everything between special comments (`//~autogen`) will be generated by the autogen system. Anything you put there by hand will be thrown out.

> Besides, there is something like this: //---

These are just usual comments used to visually separate different classes/functions.

> I just finished one of the classes (Motor) with the autogen system.

Did you actually use the autogen system, or did you just fill everything by hand? It does not look like you have the `templates` folder in your repo. Did you forget to commit it?

mob41, 2016-04-01:
@ddemidov  Sorry for late reply.

I am having problems with the autogen system in my VM. It keeps jumping this error:
```shell
anthonylaw@Anthony-DevVM:~/dev/ev3dev-lang-java/ev3dev-lang$ nodejs ./exec-autogen.js ../../autogen-config.js

module.js:340
    throw err;
          ^
Error: Cannot find module 'yargs'
    at Function.Module._resolveFilename (module.js:338:15)
    at Function.Module._load (module.js:280:25)
    at Module.require (module.js:364:17)
    at require (module.js:380:17)
    at Object.<anonymous> (/home/anthonylaw/dev/ev3dev-lang-java/ev3dev-lang/autogen/autogen.js:6:12)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Module.require (module.js:364:17)
```

<b>System Version:</b> Lubuntu 15.10
<b>Kernel Version:</b> Linux 4.2.0-16-generic (i686)
<b>Desktop:</b> LXDE (Lubuntu)

<b>Node.js Version:</b> v0.10.25
<b>NPM Version: </b> 1.4.21

ddemidov, 2016-04-01:
> *System Version:* Lubuntu 15.10
> *Kernel Version:* Linux 4.2.0-16-generic (i686)
> *Desktop:* LXDE (Lubuntu)
> *Node.js Version:* v0.10.25
> *NPM Version: * 1.4.21

Cheers,
Denis

ddemidov, 2016-04-01:
Did you do `npm install` inside `ev3dev-lang`?

mob41, 2016-04-01:
Opps, missing.

mob41, 2016-04-01:
...whatever, it throws me another error:
```
anthonylaw@Anthony-DevVM:~/dev/ev3dev-lang-java/ev3dev-lang$ npm install
npm ERR! install Couldn't read dependencies
npm ERR! package.json ENOENT, open '/home/anthonylaw/dev/ev3dev-lang-java/ev3dev-lang/package.json'
npm ERR! package.json This is most likely not a problem with npm itself.
npm ERR! package.json npm can't find a package.json file in your current directory.

npm ERR! System Linux 4.2.0-16-generic
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install"
npm ERR! cwd /home/anthonylaw/dev/ev3dev-lang-java/ev3dev-lang
npm ERR! node -v v0.10.25
npm ERR! npm -v 1.4.21
npm ERR! path /home/anthonylaw/dev/ev3dev-lang-java/ev3dev-lang/package.json
npm ERR! code ENOPACKAGEJSON
npm ERR! errno 34
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/anthonylaw/dev/ev3dev-lang-java/ev3dev-lang/npm-debug.log
npm ERR! not ok code 0
```

I also tried with root permissions, doesn't work too.

ddemidov, 2016-04-01:
My bad, `npm install` should be run inside `ev3dev-lang/autogen`.

mob41, 2016-04-01:
Opps, I found that I cloned the ```develop``` branch. Should I clone the ```master``` branch instead?

ddemidov, 2016-04-01:
`develop` is the preferred branch I think.

mob41, 2016-04-01:
Nope, it doesn't work.

mob41, 2016-04-01:
The log is too long. But I figured out the problem is that the script is pointing ```nodejs``` command as ```node```!
I tried to do ```sudo apt-get install node```, but the package ```nodejs``` was suggested though.

mob41, 2016-04-01:
(No installation candidate)

ddemidov, 2016-04-01:
We need an advice from @WasabiFan then, I am out of ideas. Btw, on my machine `/usr/bin/node` is installed by nodejs package.

mob41, 2016-04-01:
Maybe make a symlink?

mob41, 2016-04-01:
There is another package called ```nodejs-legacy```, which is not recommended, I think. Maybe the script should be edited for Ubuntu 15.10 as ```wily``` changed a lot of things from 14.04

mob41, 2016-04-01:
I just installed it. The command ```node``` is able to use again. I am trying with ```npm install``` again

WasabiFan, 2016-04-01:
`cd` into the autogen directory within ev3dev-lang/ and run `npm install` from there. Sorry -- I discovered this issue earlier this week and forgot to document it.

mob41, 2016-04-02:
I don't have enough time now, as I need to follow with my school's tasks. Whatever, I forget to make the liquid templates files to the ```templates``` folder.

mob41, 2016-04-04:
I have another problem that the unified libraries in the [http://www.ev3dev.org/docs/libraries/](ev3dev site). They are called "official unified libraries". Is there only 3 libraries currently maintained by the official ev3dev programmers such as @ddemidov , @dlech , @rhempel ?

closed 2016-04-04 by mob41

ddemidov, 2016-04-04:
The libraries are 'official' in the sense that they follow the unified specification in this repo. Yes, there are only three of those currently. I think only @dlech and @rhempel may be called 'official' ev3dev programmers, but, like the rest of contributors, they are doing this for fun, not for profit.

So, what is the exact problem here?

mob41, 2016-04-04:
"What should I do after writing the unified library? Should I modify the webpage myself and make a pull request?"

ddemidov, 2016-04-04:
Yes, that would be fine.

rhempel, 2016-04-04:
First, @dlech and @rhempel also do this for fun, there is no funding or financial aid coming in to this project at all.

Next, if you're making what you hope will be an "official" binding for Java that follows the language spec it does not have to use the `autogen` system, but your work will be simplified later if we change the spec at all.

Since we do not yet have an official Java binding, I suggest that you host it in your own GitHub space and notify us that it's ready for review.

WasabiFan, 2016-04-04:
@mob41 Sorry if this seems a bit disorganized! Yes, once you have gotten your library ready feel free to tell us and open a PR on the site to add it to the list.

jabrena, 2016-04-04:
@rhempel In relation to a official Java Binding, I think that in the next release of the kernel, I could add the library. This was the purpose of the library:
https://github.com/ev3dev-lang-java/ev3dev-lang-java

I would like to clarify this point.

Juan Antonio