ev3dev/ev3dev-lang #157: Unified language for Java submission

mob41, 2016-04-11:
Hi,

I just completed the unified language library for ev3dev.

Here's the downloads, wiki and JavaDoc:
[[Unstable release Build 98]](https://github.com/mob41/ev3dev-lang-java/releases/tag/unstable-0.0.1-SNAPSHOT-B98)
[[Wiki]](https://github.com/mob41/ev3dev-lang-java/wiki)
[[JavaDoc]](https://mob41.github.io/ev3dev-lang-java/)

WasabiFan, 2016-04-11:
Thanks! I'll take a look when I get some time.

Whenever we get to adding this to our language lists, I'll need to remember to explain both this and @jabrena's library and how they are different.

WasabiFan, 2016-04-24:
Sorry, I haven't forgotten about this -- just postponed it while I got other stuff done :wink: It's tough maintaining projects on GitHub when you have "real life work" to do!

To make sure I understand, what do you want us to do as a result of this issue other than adding this library to our lists in the README and on the site?

mob41, 2016-04-26:
@WasabiFan Sorry for waiting for a long time for a reply. I am so, so, busy on school projects and tasks.

I don't completely understand what you mainly asking about. Are you asking me to add this to the site list and the README?

WasabiFan, 2016-04-26:
> Sorry for waiting for a long time for a reply. I am so, so, busy on school projects and tasks.

No problem! I'm in a very similar situation myself.

> I don't completely understand what you mainly asking about. Are you asking me to add this to the site list and the README?

I was just asking what you wanted me to do. I assumed that your intent with this issue was for me to add your library to the list of "official ones" on the README and on the site, but I wanted to confirm that that was what you wanted. Is that what you want me to do (meaning that your library is ready to be promoted)? Or was there something else that you wanted us to do?

mob41, 2016-04-27:
...just for quick reply

Yup, you can just add the library to the list.
Besides, I will go and update the library to support kernel v10 in these
few days. Monday is holiday, yeah.

mob41, 2016-04-27:
The library was updated to v10 now...

closed 2017-10-06 by mob41