ev3dev/ev3dev-lang #158: Having issue to connect Pycharm to the Brick

Snooone, 2016-04-29:
Hi all,

This is my first attempt to code lego bricks using ev3dev. I am connecting from a Win 10 PC. 

As per the details documented (great job!) in ev3dev, I managed to connect through USB, then through WIFI. 
Putty is connecting well. And I launched the apt-get update ... upgrade, dist-upgrade, reboot and so on.

But now I am stucked, Pycharm on windows cannot connect the brick, it triggers  this error:

       Can't login: Remote host closed connnection during hand shake

I am using following details:
   Host: robot@192.168.0.47/myproject
   login: robot
   password: maker

local brick git directory: myproject.git

has someone already experience this kind of issue?

Thanks & regards
Serge

WasabiFan, 2016-04-29:
Glad to hear that you were able to get the brick mostly up-and-running! Can you reopen this issue on [the main ev3dev repo](https://github.com/ev3dev/ev3dev)? This repo is for the programming libraries specifically, but this problem seems to be with ev3dev in general and not the libraries. When you open the issue there, it should automatically have an "issue template" that asks for some extra information -- make sure that you fill that out.

Snooone, 2016-04-29:
Hi Was, sorry for the mistake, I will post on the other trail.
Thanks

closed 2016-04-29 by Snooone

WasabiFan, 2016-04-29:
No problem! I just want to make sure that your question gets to the right place so that you can get responses from the right people.