ev3dev/ev3dev-lang #166: Javascript motor and sensor example

liam-b, 2016-07-16:
I really need a quick demo of how to use node.js to run motors and get inputs from sensors. I've got npm and node installed and I've required the language like so:
``` javascript
var ev3dev = require('ev3dev-lang');

var motor = new ev3dev.LargeMotor('outA'); // trying to setup a new motor
```

but I can't find any code examples anywhere online. Can someone show me how to get anything working? Thanks

WasabiFan, 2016-07-16:
Check out the [examples directory](https://github.com/WasabiFan/ev3dev-lang-js/tree/master/examples) in the ev3dev-lang-js repo; there you can find examples on how to use motors, sensors and various other things. You should mainly check out these ones:
- [run-specific-motor.js](https://github.com/WasabiFan/ev3dev-lang-js/blob/master/examples/run-specific-motor.js)
- [led-from-motor-speed.js](https://github.com/WasabiFan/ev3dev-lang-js/blob/master/examples/led-from-motor-speed.js)
- [touch-sensor-motor-control.js](https://github.com/WasabiFan/ev3dev-lang-js/blob/master/examples/touch-sensor-motor-control.js)

WasabiFan, 2016-07-16:
Also, make sure to check out the online documentation. The motor methods are [here](http://wasabifan.github.io/ev3dev-lang-js/classes/_motors_.motor.html#runfordistance).

liam-b, 2016-07-25:
Hey thanks, that really helps :D

closed 2016-07-25 by WasabiFan