ev3dev/ev3dev-lang #174: Some typo on the driver docs

mob41, 2016-10-12:
Hi,

I found out some typo in the following document: http://www.ev3dev.org/docs/sensors/hitechnic-nxt-color-sensor/ , whereas ```vendor_id``` would be ```HiTechnc``` instead of ```HiTechnic```.

I am not quite sure about it, but I think it is ```HiTechnic```

Regards,
mob41

ddemidov, 2016-10-12:
`HiTechnc` is consistently used as a `vendor_id` throughout this file (it is used to generate the page you linked): https://github.com/ev3dev/lego-linux-drivers/blob/master/sensors/nxt_i2c_sensor_defs.c. So this could be the intended value.

ddemidov, 2016-10-12:
Also, `vendor_id` is [defined](https://github.com/ev3dev/lego-linux-drivers/blob/master/sensors/nxt_i2c_sensor_core.c#L400) as `char[NXT_I2C_ID_STR_LEN + 1]`, where [`NXT_I2C_ID_STR_LEN` is 8](https://github.com/ev3dev/lego-linux-drivers/blob/master/sensors/nxt_i2c_sensor.h#L25). So `HiTechnic` simply does not fit into `vendor_id`

mob41, 2016-10-18:
Just a suggestion: Can this be told at the docs? So the users won't be confused with HiTechnc