ev3dev/ev3dev-lang #175: stop lcd going back to menu

absentdream, 2016-12-23:
hi 
i am using bottle (web server) to remote control ev3 but when i update the screen image it reverts back to menu after 1 sec (ish)... 

`from bottle import route, request, run, template
import requests
import time
import ev3dev.ev3 as ev3
from PIL import Image, ImageDraw, ImageFont, ImageOps

@route('/motor/<name>/timed/<time_sp>/<speed_sp>')
def index(name,time_sp,speed_sp):
	m = ev3.LargeMotor(name)
	m.run_timed(time_sp=time_sp, speed_sp=speed_sp)
	while 'running' in m.state: time.sleep(0.01)
	return template('<b>motor</b>  name: {{name}} timed', name=name)
	
@route('/eyes/<name>')
def index(name):
	logo = Image.open('data/images/' + name + '.bmp')
	lcd = ev3.Screen()
	lcd.draw.bitmap((0, 0), ImageOps.invert(logo.convert("L")))
	lcd.update()
	return template('<b>eyes</b>  name: {{name}}', name=name)

run(host='192.168.0.33', port=8080)`

any ideas how to stop it reverting to menu..

WasabiFan, 2016-12-23:
The problem is that Brickman (the program that displays the menu) still thinks it has control of the display. If you use the on-screen menu to run your script, that should solve the problem, because that means that Brickman knows that another program is using the display. Otherwise, see the bottom of the page here: https://sites.google.com/site/ev3python/learn_ev3_python/screen

absentdream, 2016-12-23:
spot on thanks.. using "sudo chvt 6" to switch screen ...

closed 2016-12-23 by absentdream