ev3dev/ev3dev-lang #179: UNMET PEER DEPENDENCY bluebird@~3.5.0

dwalton76, 2017-09-09:
Deleted node_modules and re-ran `npm install` but I get an error about bluebird
```
dwalton@laptop ~/l/e/e/autogen> sudo rm -rf node_modules/
dwalton@laptop ~/l/e/e/autogen>
dwalton@laptop ~/l/e/e/autogen> tree .
.
├── autogen-core.js
├── autogen.js
├── config.js
├── package.json
├── README.rst
└── utils.js

0 directories, 6 files
dwalton@laptop ~/l/e/e/autogen> 
dwalton@laptop ~/l/e/e/autogen> sudo npm install

> liquid-node@2.4.0 install /home/dwalton/lego/ev3dev-lang-python/ev3dev-lang/autogen/node_modules/liquid-node
> npm run compile

> liquid-node@2.4.0 compile /home/dwalton/lego/ev3dev-lang-python/ev3dev-lang/autogen/node_modules/liquid-node
> node ./npm-compile.js

npm WARN lifecycle ev3dev-lang-autogen@1.0.0~postinstall: cannot run in wd %s %s (wd=%s) ev3dev-lang-autogen@1.0.0 cd node_modules/liquid-node && npm install && npm run compile /home/dwalton/lego/ev3dev-lang-python/ev3dev-lang/autogen
ev3dev-lang-autogen@1.0.0 /home/dwalton/lego/ev3dev-lang-python/ev3dev-lang/autogen
├── UNMET PEER DEPENDENCY bluebird@~3.5.0
├─┬ liquid-node@2.4.0  (git://github.com/WasabiFan/liquid-node.git#ed542e502c30c09ff234f49c6ef1757b613d8168)
│ └── strftime@0.8.4 
└─┬ yargs@3.29.0 
  ├── camelcase@1.2.1 
  ├─┬ cliui@3.2.0 
  │ ├─┬ string-width@1.0.2 
  │ │ ├── code-point-at@1.1.0 
  │ │ └─┬ is-fullwidth-code-point@1.0.0 
  │ │   └── number-is-nan@1.0.1 
  │ ├─┬ strip-ansi@3.0.1 
  │ │ └── ansi-regex@2.1.1 
  │ └── wrap-ansi@2.1.0 
  ├── decamelize@1.2.0 
  ├─┬ os-locale@1.4.0 
  │ └─┬ lcid@1.0.0 
  │   └── invert-kv@1.0.0 
  ├── window-size@0.1.4 
  └── y18n@3.2.1 

npm WARN liquid-node@2.4.0 requires a peer of bluebird@~3.5.0 but none was installed.
npm WARN ev3dev-lang-autogen@1.0.0 No repository field.
npm WARN ev3dev-lang-autogen@1.0.0 No license field.
dwalton@laptop ~/l/e/e/autogen> 
```

Running `node autogen-config.json` from the ev3dev-lang-python root directory doesn't make any changes
```
dwalton@laptop ~/l/ev3dev-lang-python> node autogen-config.json 
dwalton@laptop ~/l/ev3dev-lang-python> git diff
diff --git a/ev3dev-lang b/ev3dev-lang
--- a/ev3dev-lang
+++ b/ev3dev-lang
@@ -1 +1 @@
-Subproject commit 008804f5c28acd225f1b43057be2060d9965b2c1
+Subproject commit 008804f5c28acd225f1b43057be2060d9965b2c1-dirty
dwalton@laptop ~/l/ev3dev-lang-python> 
```

WasabiFan, 2017-09-09:
@dwalton76 Try again; delete `node_modules` and re-run `npm install` from the autogen directory.

Once that succeeds, the command to run autogen will be `node ev3dev-lang/autogen/autogen.js`.

WasabiFan, 2017-09-09:
By the way, I _am_ actually changing things in one of the dependencies each time I tell you to try again -- I'm not just sending you in circles :laughing: This time I actually switched over to an Ubuntu machine and fixed things from there.

dwalton76, 2017-09-10:
`npm install` now appears to be ok
```
dwalton@laptop ~/l/e/e/autogen> sudo rm -rf node_modules/
dwalton@laptop ~/l/e/e/autogen> sudo npm install

> liquid-node@2.4.0 install /home/dwalton/lego/ev3dev-lang-python/ev3dev-lang/autogen/node_modules/liquid-node
> npm run compile

> liquid-node@2.4.0 compile /home/dwalton/lego/ev3dev-lang-python/ev3dev-lang/autogen/node_modules/liquid-node
> nodejs ./npm-compile.js

npm WARN lifecycle ev3dev-lang-autogen@1.0.0~postinstall: cannot run in wd %s %s (wd=%s) ev3dev-lang-autogen@1.0.0 cd node_modules/liquid-node && npm install && npm run compile /home/dwalton/lego/ev3dev-lang-python/ev3dev-lang/autogen
ev3dev-lang-autogen@1.0.0 /home/dwalton/lego/ev3dev-lang-python/ev3dev-lang/autogen
├─┬ liquid-node@2.4.0  (git://github.com/WasabiFan/liquid-node.git#5018c3b27852646b1defdddeaf23e2f0db4e63f5)
│ ├── bluebird@3.5.0 
│ └── strftime@0.8.4 
└─┬ yargs@3.29.0 
  ├── camelcase@1.2.1 
  ├─┬ cliui@3.2.0 
  │ ├─┬ string-width@1.0.2 
  │ │ ├── code-point-at@1.1.0 
  │ │ └─┬ is-fullwidth-code-point@1.0.0 
  │ │   └── number-is-nan@1.0.1 
  │ ├─┬ strip-ansi@3.0.1 
  │ │ └── ansi-regex@2.1.1 
  │ └── wrap-ansi@2.1.0 
  ├── decamelize@1.2.0 
  ├─┬ os-locale@1.4.0 
  │ └─┬ lcid@1.0.0 
  │   └── invert-kv@1.0.0 
  ├── window-size@0.1.4 
  └── y18n@3.2.1 

npm WARN ev3dev-lang-autogen@1.0.0 No repository field.
npm WARN ev3dev-lang-autogen@1.0.0 No license field.
dwalton@laptop ~/l/e/e/autogen> 
```

 but `node autogen-config.json` still doesn't make any changes to core.py, etc
```
dwalton@laptop ~/l/e/e/autogen> pwd
/home/dwalton/lego/ev3dev-lang-python/ev3dev-lang/autogen
dwalton@laptop ~/l/e/e/autogen> cd ../..
dwalton@laptop ~/l/ev3dev-lang-python> 
dwalton@laptop ~/l/ev3dev-lang-python> node autogen-config.json 
dwalton@laptop ~/l/ev3dev-lang-python> git diff
diff --git a/ev3dev-lang b/ev3dev-lang
--- a/ev3dev-lang
+++ b/ev3dev-lang
@@ -1 +1 @@
-Subproject commit 008804f5c28acd225f1b43057be2060d9965b2c1
+Subproject commit 008804f5c28acd225f1b43057be2060d9965b2c1-dirty
dwalton@laptop ~/l/ev3dev-lang-python> 
```

dwalton76, 2017-09-10:
The changes I have in spec.json that I am trying to get applied to core.py
```
dwalton@laptop ~/l/ev3dev-lang-python> git diff
diff --git a/ev3dev-lang b/ev3dev-lang
--- a/ev3dev-lang
+++ b/ev3dev-lang
@@ -1 +1 @@
-Subproject commit 008804f5c28acd225f1b43057be2060d9965b2c1
+Subproject commit 008804f5c28acd225f1b43057be2060d9965b2c1-dirty
dwalton@laptop ~/l/ev3dev-lang-python> cd ev3dev-lang/
dwalton@laptop ~/l/e/ev3dev-lang> git diff
diff --git a/spec.json b/spec.json
index 0891772..ea708c2 100644
--- a/spec.json
+++ b/spec.json
@@ -1482,10 +1482,10 @@
         "ev3": {
             "led": {
                 "instances": [
-                    { "name" : "Red Left", "systemName": "ev3:left:red:ev3dev" },
-                    { "name" : "Red Right", "systemName": "ev3:right:red:ev3dev" },
-                    { "name" : "Green Left", "systemName": "ev3:left:green:ev3dev" },
-                    { "name" : "Green Right", "systemName": "ev3:right:green:ev3dev" }
+                    { "name" : "Red Left", "systemName": "led0:red:brick-status" },
+                    { "name" : "Red Right", "systemName": "led1:red:brick-status" },
+                    { "name" : "Green Left", "systemName": "led0:green:brick-status" },
+                    { "name" : "Green Right", "systemName": "led1:green:brick-status" }
                 ],
                 "groups" : [
                     { "name" : "Left", "entries" : ["Red Left", "Green Left"] },
dwalton@laptop ~/l/e/ev3dev-lang> 
```

WasabiFan, 2017-09-10:
You can't execute a JSON file; see my comment above for the command to run.

dwalton76, 2017-09-10:
doh!! :) working now...thank you @WasabiFan 
```
dwalton@laptop ~/l/ev3dev-lang-python> node ev3dev-lang/autogen/autogen.js
Completed processing file "/home/dwalton/lego/ev3dev-lang-python/ev3dev/brickpi.py"
Completed processing file "/home/dwalton/lego/ev3dev-lang-python/ev3dev/core.py"
Completed processing file "/home/dwalton/lego/ev3dev-lang-python/spec_version.py"
Completed processing file "/home/dwalton/lego/ev3dev-lang-python/docs/sensors.rst"
Completed processing file "/home/dwalton/lego/ev3dev-lang-python/ev3dev/ev3.py"
dwalton@laptop ~/l/ev3dev-lang-python> 
dwalton@laptop ~/l/ev3dev-lang-python> git diff
diff --git a/ev3dev-lang b/ev3dev-lang
index 008804f..31af6d1 160000
--- a/ev3dev-lang
+++ b/ev3dev-lang
@@ -1 +1 @@
-Subproject commit 008804f5c28acd225f1b43057be2060d9965b2c1
+Subproject commit 31af6d13b7ee1ee06c1a9ecfb54f62f904770b8a-dirty
diff --git a/ev3dev/ev3.py b/ev3dev/ev3.py
index 725f713..8639784 100644
--- a/ev3dev/ev3.py
+++ b/ev3dev/ev3.py
@@ -47,10 +47,10 @@ class Leds(object):
 
 # ~autogen led-colors platforms.ev3.led>currentClass
 
-    red_left = Led(name_pattern='ev3:left:red:ev3dev')
-    red_right = Led(name_pattern='ev3:right:red:ev3dev')
-    green_left = Led(name_pattern='ev3:left:green:ev3dev')
-    green_right = Led(name_pattern='ev3:right:green:ev3dev')
+    red_left = Led(name_pattern='led0:red:brick-status')
+    red_right = Led(name_pattern='led1:red:brick-status')
+    green_left = Led(name_pattern='led0:green:brick-status')
+    green_right = Led(name_pattern='led1:green:brick-status')
 
     LEFT = ( red_left, green_left, )
     RIGHT = ( red_right, green_right, )
```

I will submit a PR here for the led change

closed 2017-09-10 by dwalton76

WasabiFan, 2017-09-10:
Haha, no problem :laughing: Our script depends on a forked version of a module which was missing an upstream fix, I just had to apply the proper bits of that and everything worked.