ev3dev/ev3dev-lang #182: Buttons Error (platform-gpio-keys.0-event not found)

laurensvalk, 2017-09-24:
It appears that Buttons do not work.

I'm using the [latest ev3dev stretch on my ev3](https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/2017-09-14/snapshot-ev3dev-stretch-ev3-generic-2017-09-14.img.xz), with no modifications/additional installs.

Possibly similar to [this](https://github.com/rhempel/ev3dev-lang-python/issues/193), but the error looks different. 

```
Traceback (most recent call last):
  File "/home/robot/testing/prog.py", line 4, in <module>
    btn = ev3.Button()
  File "/usr/lib/python3/dist-packages/ev3dev/core.py", line 2554, in __init__
    self._file_cache[name] = open(name, 'rb', 0)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/input/by-path/platform-gpio-keys.0-event'
```

Minimum example, [copied from here](https://github.com/rhempel/ev3dev-lang-python/issues/193#issuecomment-244627313).

```
#!/usr/bin/env python3
import ev3dev.ev3 as ev3

btn = ev3.Button()
# Is 'Left' button pressed?
print('yes' if btn.left else 'no')

# Check if any buttons are pressed:
print('yes' if btn.any() else 'no')
```

laurensvalk, 2017-09-24:
I've posted this in the wrong place. I'll repost in ev3dev-lang-python.

closed 2017-09-24 by laurensvalk

ddemidov, 2017-09-24:
This should be fixed by #181 (but is not yet published).

laurensvalk, 2017-09-24:
Ok, great!