ev3dev/ev3dev-lang #183: HiTechnic Compass Sensor "Device is not connected"

CZDzen, 2018-02-20:
Kernel version 4.4.87
code:
```
#!/usr/bin/env python3

from ev3dev.ev3 import *
from time import sleep
mB = LargeMotor('outB')
mC = LargeMotor('outC')
comp = Sensor(address='in1:i2c1', driver='ht-nxt-compass')
t = 0
while t<100:
        print (comp.value())
        sleep(1)
        t +=1
```
Error:
Traceback (most recent call last):
  File "./forward", line 12, in <module>
    print (comp.value())
  File "/usr/lib/python3/dist-packages/ev3dev/core.py", line 1758, in value
    self._value[n], value = self.get_attr_int(self._value[n], 'value'+str(n))
  File "/usr/lib/python3/dist-packages/ev3dev/core.py", line 214, in get_attr_int
    attribute, value = self._get_attribute(attribute, name)
  File "/usr/lib/python3/dist-packages/ev3dev/core.py", line 198, in _get_attribute
    raise Exception('Device is not connected')
Exception: Device is not connected

dlech, 2018-02-20:
Does the sensor show up in BrickMan under Device Browser?

Also it would be better if the bug report was filed in the ev3dev-lang-python repository.

https://github.com/ev3dev/ev3dev-lang-python/issues

CZDzen, 2018-02-21:
> Does the sensor show up in BrickMan under Device Browser?

Yes, i see him in in1:i2c1 port and i can watch values

dlech, 2018-02-21:
OK, so not a driver issue. More python people will see your issue if you post it at: https://github.com/ev3dev/ev3dev-lang-python/issues

closed 2025-04-02 by SchwarzFuchs